Tow-Boot/Tow-Boot #18: Can't boot spi-installer on pinebook pro

jbedo, 2021-05-19:
I'm trying to install NixOS on my PBP and I've built both Tow-Boot and your NixOS sdimage (the new feature branch). I can't seem to boot the spi-installer.img though; after writing it to the sd card the PBP never boots from it. The sdimage produced by your overlay works fine otoh, so I'm guessing it's an issue with the spi-installer.img. Since my PBP boots currently, can I just write the SPI image directly on the PBP instead of booting from the image?

samueldr, 2021-05-19:
> I can't seem to boot the spi-installer.img though

What do you mean from that? On a Pinebook Pro, the *initial boot firmware* selected will be, in order, the one in SPI, the one on eMMC or the one on SD. So if you have U-Boot on SPI, the *IBF* installed on the SD card will never itself boot.

But, that doesn't mean the SD installer can't work. If you select, somehow, to boot from SD, it *should* be able to do *something* with it. Hopefully present you with the `bootmenu`-based menu to erase/install. Though maybe there's a bug... The `spi-installer.img` as released might not work correctly depending on what U-Boot booted it. Currently it has been mainly tested against the same revision Tow-Boot build. It is something to work on, actually test and use the SPI installer outside of Tow-Boot so a user can upgrade to Tow-Boot.

* * *

Now for other methods of installing:

Do you have a U-Boot installed to the SPI Flash? You could erase it so the Tow-Boot on SD is used, which in turn should be able to "boot itself".

Alternatively, you *can* "just" write `firmware.spiflash.bin` to the SPI Flash. I don't know how, haven't investigated yet, but it is possible on Linux.

Otherwise, with U-Boot the important commands are those:

https://github.com/Tow-Boot/Tow-Boot/blob/a4a624473c59b1c3c6d16ed19de11fb8baeac343/support/builders/spi-installer/default.nix#L23-L26

Where mainly, it loads the firmware binary, the `if` ensures the script fails if loading fails.

Then, `sf probe` makes the SPI flash device available, `sf erase` is more of a formality, I'm not 100% positive it is strictly required. Finally `sf write` well... it writes the firmware! But if you were to just run those commands, I would invite you to *instead* just `sf erase` and then boot the SD card SPI installer. This way you can ensure the Tow-Boot version there should work.

jbedo, 2021-05-20:
Thanks, with those pointers I managed to boot from the SD after erasing the SPI. Tow-Boot installed fine and can boot itself too. Thanks for the help!

closed 2021-05-20 by jbedo

RiderExMachina, 2022-03-24:
I ran into this issue as well with the December 2021 U-boot release. I ended up using `nandwrite` to write the `binaries/Tow-Boot.spi.bin` file from the 2021.10-004 release. For anyone who was just as worried as I was, it did flash successfully after writing 379 blocks.

kkeijzer, 2022-04-16:
Same problem here. The SD card also wouldn't boot, but I could run `sudo nandwrite -p /dev/mtd0 Tow-Boot.spi.bin` from Linux, and after writing 379 blocks Tow-Boot was installed.

On Debian/Ubuntu, `nandwrite` is part of the `mtd-utils` package.

Maybe this should become an official instruciton?

bburdette, 2022-06-12:
I installed nandwrite, but I don't have a `/dev/mtd0` device.  How do I enable that?

pjsfx, 2022-06-14:
I've just been going through this too. on an original october 2019 pbp there does appear to be something in the spi from the factory, and the original mr fixit debian doesn't expose /dev/mtd0 even with mtd-utils  installed. what I've done is to install current manjaro to an sd card and boot from that. manjaro does show /dev/mtd0 so you can install mtd-utils  (with sudo pacman -S mtd-utils) and do sudo nandwrite -p /dev/mtd0 Tow-Boot.spi.bin from there. Unfortunately for me I now can't boot anything but the emmc since the keyboard in my pbp is broken and the tow-boot I installed is defaulted to boot emmc first doesn't seem to recognise usb keyboards.

CameronNemo, 2022-09-08:
`/dev/mtd0` is provided by the kernel. If it is not showing up, then you are missing a kernel configuration. I am not 100% sure, but I think the config needed is `CONFIG_SPI_ROCKCHIP=y` for the rk3399.

Paulie420, 2022-09-23:
Just so others know, the direct writing way worked for me too;
dd if=Tow-boot.spi.bin of=/dev/mtd0

(/dev/mtd0 may not be listed in lsblk, but you prolly still have it in /dev....) Thanks to the original poster mentioning this route. Tow Boot on Pinebook Pro worked fine.

vbmithr, 2022-10-15:
Can't boot from SD spi installer either for RockPro64 here.

River-b, 2022-11-25:
> Thanks, with those pointers I managed to boot from the SD after erasing the SPI. Tow-Boot installed fine and can boot itself too. Thanks for the help!

what was exact solutions please?  I've been struggling with this for many days now.  I can boot to SD Armbian.   And I  even Deleted  the   UBOOT from emmc  manjaro (which is why manjaro will not boot). 

-- I also deleted the SPI  using   /dev/mtd0 000

River-b, 2022-11-25:
> Just so others know, the direct writing way worked for me too;
> dd if=Tow-boot.spi.bin of=/dev/mtd0

I tried this too from within  Manjaro when I still had it working and from within the SD  Armbian boot up. 
Nothing happens when I reboot, no tow boot menu or icon.  I  Followed some instructions on Arch  that show how to Delete the SPI first also  and then do this install but nothing works so far. 

ARe there more additional steps you took that I might be missing?