Tow-Boot/Tow-Boot #19: Performing `saveenv` when booting from SPI corrupts SPI

b-, 2021-05-19:
This appears to be somewhat of [a u-boot problem](https://forums.xilinx.com/t5/Embedded-Linux/Where-does-U-Boot-saveenv-to/td-p/532915), and not specifically tow-boot, but I figure an opinionated configuration might provision for this sort of thing:

I had Tow-Boot installed to SPI, and I was poking around the console, and I tried running `saveenv`, which proceeded to save environment settings to SPI, overwriting part of the loader. Which part? I don't know. It didn't say _where_ in SPI it was saving, but I presume it was an important region because now the power LED is red upon power on and nothing else happens.

Maybe `saveenv` can write to a specific offset past the bootloader? I'm not sure what the best course of action is here. All I know is that I rendered my Pinebook Pro unbootable through a command with a relatively innocuous-sounding name. (Yes, I should have looked up what this did first…)

samueldr, 2021-05-19:
Oooh, that can get annoying to rescue depending on the device. Though it might be a misconfiguration on our side here. After all I had to change some settings for SPI from the defaults.

I'm currently taking time to $work on Mobile NixOS things, but will be taking some time in a not too distant future to work on Tow-Boot stuff again.

plabadens, 2021-05-19:
I have not tried any of this, so I can't vouch for these instructions, but in principle you can flash the SPI from another PC, hopefully by getting into maskrom mode using the recovery button on the mainboard. I've also heard you can force it to boot on the SD/eMMC by shorting some pins, and then erase/flash the SPI from there.

https://wiki.pine64.org/wiki/Pinebook_Pro_SPI

samueldr, 2021-05-20:
I haven't actively tried the maskrom button method, but shorting the pins for the SPI Flash works just fine.

b-, 2021-05-20:
Indeed, I actually came across Tow-Boot because of a bad SPI flash anyway. Thanks for the help, though!

samueldr, 2021-05-20:
It was stated as being equivalent to shorting the SPI pins in some documentations. But it does not work to skip SPI boot. My assumption is that there is a hardware bug that makes it not work as expected.

I can boot to maskrom mode successfully by having no valid *initial boot firmware* to boot though. So none in SPI, none in eMMC, and none in SD. I actually have to look into building Tow-Boot so that it is bootable through "maskrom" as it would facilitate development some more :).

samueldr, 2021-06-23:
If you build from #34, `saveenv` and `env save` work as expected on the boards I have tested

 - Pine A64-LTS (a64)
 - AML-S805X-AC (S805X)
 - ROC-RK3399-PC (rk3399)

Given that I never have seen a difference so fundamental between any rk3399 boards, I assume it's fixed for the Pinebook Pro. I'm unable to test just at the moment.

My testing included using `saveenv` on all those three boards. And verifying it loads the new environment.

(Side-note: By design, any non-SPI installs don't support reading/saving environment anymore.)

PR #34 (2021-06-26): Take control over the environment
This is a weird name. But saying it "fixes `saveenv` issues" is too diminutive. But hey, closes #19!

* * *

## What this changes

### Environment

Two broad strokes. For release `002` of Tow-Boot, we will **only** support environment in the SPI Flash for builds installed to the SPI Flash.

Firmware installed to a shared storage media, starting with `002` up to later will not allow saving the environment.

To reduce confusion, a Tow-Boot install will **strictly** only use the environment in the same storage it is installed into. This means that SPI-installed f [... cut]
closed 2021-06-26 by samueldr