Tow-Boot/Tow-Boot #23: Fedora generic aarch64 does not boot

b-, 2021-05-25:
Fedora fails to boot, I tried with both Workstation 34 and Rawhide, one on my eMMC and one on my microSD. (I forget which one is which).

Two screenshots: [image] [image]

It works (to an extent) when using Fedora's own uboot, package spec/source here: https://src.fedoraproject.org/fork/rdossant/rpms/uboot-tools/blob/rawhide/f/uboot-tools.spec

I know UEFI booting is incomplete, so I figure there's something either upstream or a patch in Fedora's repo that makes this work, but unfortunately I don't know enough about uboot to understand what's going on in either your nix build system or how Fedora's rpm spec works.

(Unrelated note, my own matrix server needs to be burned with fire and abandoned due to lack of maintenance; when I have a matrix home I'll be joining your matrix group).

samueldr, 2021-05-25:
Is it the UEFI installer iso, or once installed that it fails to boot?

Note that I haven't looked at the "pre-installed image" use case at all, only at using the USB iso image to do the "usual installation wizard".

Coincidentally someone yesterday on Matrix told me this:

> Fedora actually doesn't load for me because of some weird EFI error. Running `bootefi mmc 1:1 /efi/fedora/grubaa64.efi` works fine though

For which I'm also unclear whether it means the USB iso, a "pre-installed" image, or once installed from that iso.

Finally also note that I'm really not a Fedora user. My experience with Fedora starts and ends at downloading an AArch64 iso to test on some of the Tow-Boot platforms.

b-, 2021-05-25:
I'm not using a .iso on USB, but rather a raw disk image flashed to SD/eMMC.

I did miss your notes, but as I'm dedicating my SPI to Tow-Boot the only thing I see in there that seems especially relevant is that shim doesn't work. I replaced BOOTAARCH64.efi with grub from the same image and got to a boot menu and then a Fedora boot splash that eventually stopped spinning. It looks like it froze, but we're getting further...

samueldr, 2021-05-27:
*There was some more discussion on the Matrix. I don't remember what was the conclusion of that discussion though.*

b-, 2021-05-27:
I got fedora booting. I used a Rawhide Workstation image, I replaced the BOOTAA64.efi shim with the grub efi binary, and I added `blacklist=efifb efifb=off` to the command line.

I went and did something else for a few minutes after, and the power light is glowing green and I'm at a Gnome desktop. (No NVMe or WiFi, so something isn't quite right still...)

So while I wouldn't suggest closing this till more works… I can safely say that with a little light caressing Fedora _does_ boot!

Be-ing, 2022-01-30:
@b- was this with the Pinebook Pro? I am looking into using Tow Boot with Fedora on the Pinephone Pro.

samueldr, 2022-01-30:
@Be-ing we've had a user recently who apparently succesfully used the UEFI iso from Fedora without changes to *just* install with Tow-Boot present on the SPI.

So I figure whatever issues are now fixed. Anyway, since last May, I *know* that many of the underlying issues are fixed.

I am closing this issue, but @Be-ing, don't hesitate to ask here or in the matrix room if you have questions. I simply assume it will work given the success from earlier this week.

closed 2022-01-30 by samueldr