Tow-Boot/Tow-Boot #25: Document that setting the Legacy Bootable flag is needed for many distroboot boot options

samueldr, 2021-06-01:
E.g.

> If you use sgdisk, type
> ```
> sgdisk -A 3:set:2 /dev/sda
> ```
> 
> which means _set capability bit 2 (=the "legacy BIOS partition") of partition 3 of /dev/sda_.

— https://www.lightofdawn.org/wiki/wiki.cgi/BIOSBootGPT

b-, 2021-06-01:
Huh, I didn't know this was a thing. I've had trouble booting with CSM on an Intel Mac before, and I wonder if this had anything to do with it.

samueldr, 2021-06-01:
If it was, it was unrelated to this specific problem. (Being that CSM and UEFI and Intel Mac are not U-Boot platforms :))

Though yeah it's plausible that an "old-school" legacy boot bootloader (booted through CSM) looks at the "bootable" flag, which in GPT has to be set in a different manner.

b-, 2021-06-01:
Indeed, I just didn't even know there was a "legacy bootable" flag in GPT. I should read more about it

samueldr, 2021-06-01:
In theory if the software was unaware of GPT and the system was using an ["Hybrid MBR"](https://en.wikipedia.org/wiki/GUID_Partition_Table#Hybrid_MBR_(LBA_0_+_GPT)) it wouldn't use the legacy flag, it would read the bootable flag on the MBR.

Though that fact only serves to muddy the waters some more :).