Tow-Boot/Tow-Boot #29: Menu pollutes environment, preventing boot commands from running

Thra11, 2021-06-06:
If left uninterrupted, it will boot correctly. However, if I press ctrl+C at the appropriate moment to enter the tow-boot menu, none of the boot options work. If I enter the "firmware console", none of the usual commands (such as `run distro_bootcmd` or `run bootcmd_mmc0`) work correctly.

If I run `env default -a` first, then `run distro_bootcmd`or `run bootcmd_mmc0` will work, confirming that something in the environment is being set in a way that prevents it booting. When I have time, I'll print the env over serial before and after resetting to defaults, and try to narrow down the culprit.

Thra11, 2021-06-07:
After choosing "Firmware console" and running `env default -a`, running `run menucmd`, then choosing "Boot from SD" also works.

Thra11, 2021-06-19:
Initially assumed that this issue would be device-agnostic. However, from the devices I've tested, only the Olimex Teres I fails. The pinebook pro doesn't appear to be affected by this issue.

samueldr, 2021-06-20:
I'm planning to work on Tow-Boot this week (the one starting in less than two days).

Have you tried other Allwinner A64 platforms?

I still haven't tried with any of those I have around.

But anyway I do recall something *similar* happening on at least another platform, so I don't doubt the issue exists. I sure hope it's nothing specific to the Teres I :)

SvenKiljan, 2021-06-20:
Can confirm that this issue occurs **sometimes** on the Pinebook Pro. After confirming the boot option with ENTER, the menu simply hangs.

This is quite rare, though. With a rough estimation, I think it works 9 out of 10 times.

samueldr, 2021-06-20:
I would hazard a guess that the hang is unrelated to this particular issue. The way I assume the environmental pollution fails on @Thra11's particular board wouldn't happen intermittently.

@SvenKiljan can you get the serial output for such a hang? And also tell what you are attempting to boot, on what kind of media?

SvenKiljan, 2021-06-21:
@samueldr 

Unfortunately, I was unable to reproduce the issue. Note that I experimented with different ways of running Tow-Boot, and it might have been a fluke.

If it happens again, I will definitely forward you the serial output.

Oh, and Tow-Boot is installed by default in [Arch Linux ARM for Pinebook Pro](https://github.com/SvenKiljan/archlinuxarm-pbp). I'm looking forward to future updates. 👍🏻

samueldr, 2021-06-21:
@Thra11 you were building against the tip of development, right?

Tip that includes the following?

 - https://github.com/Tow-Boot/Tow-Boot/commit/5ec6abbe8c85758084d27d0a5c3179e08d52a978

Uh... I guess it must otherwise the Teres I isn't available...

* * *

I've been testing with the Pinebook A64, by taking the Teres I defconfig and editing it just enough so that the Pinebook A64 works, and I can't seem to reproduce the issue... BUT, I'm testing against the internal eMMC which has a GPT+ESP with GRUB. Maybe it somehow doesn't affect that combination.

Another thing I am thinking is that maybe it is loading the environment from somewhere?

E.g. here I only have:

```
Loading Environment from FAT... *** Warning - bad CRC, using default environment
```

Look through your logs for `Loading Environment from`. AFAIUI it should always be printed.

Thra11, 2021-06-22:
Sorry I didn't get back to you before the weekend. Yes, I was testing the teres PR (#26) as-is (not rebased or anything).

I checked the env today. The differences (going from the default env to the polluted env) are as follows:
```diff
33a32
> fdtcontroladdr=b9f2ac60
38a38
> mmc_bootdev=1
48a49
> serial#=92c000ba7bd5c66f
53a55
> target_name=DHCP
61a64
> usb_pgood_delay=2000
65c68
< Environment size: 5262/131068 bytes
---
> Environment size: 5363/131068 bytes
```

> I can't seem to reproduce the issue... BUT, I'm testing against the internal eMMC

I've actually switched to booting from the internal eMMC since I reported this, and it doesn't actually appear to be affected by this issue. I might need to boot from and SD card to confirm the differences...

> ```
> Loading Environment from FAT... *** Warning - bad CRC, using default environment
> ```
> Look through your logs for Loading Environment from. AFAIUI it should always be printed.

I don't think it'll be printed if there is a valid environment to read. e.g. if you run `env save` (either with the default environment or after making modifications), then it won't complain about the bad CRC on subsequent boots. I haven't done that, so it is a little surprising that it doesn't get printed.

Attached:
| File | |
------ | ------
| `teres-boot-output.txt` | Complete serial output from u-boot |
| `teres-boot-output-before.txt
[teres-boot-output-after.txt](https://github.com/Tow-Boot/Tow-Boot/files/6696613/teres-boot-output-after.txt)
[teres-boot-output-before.txt](https://github.com/Tow-Boot/Tow-Boot/files/6696614/teres-boot-output-before.txt)
[teres-boot-output.txt](https://github.com/Tow-Boot/Tow-Boot/files/6696615/teres-boot-output.txt)
` | `env print` after pressing esc to enter the menu, then choosing firmware console, _before_ running `env default -a` |
| `teres-boot-output-after.txt` | `env print` _after_ running `env default -a` |

samueldr, 2021-06-22:
> I don't think it'll be printed if there is a valid environment to read. e.g. if you run env save (either with the default environment or after making modifications), then it won't complain about the [...]

It [will show `Loading Environment from %s...`](https://source.denx.de/u-boot/u-boot/-/blob/1ce892cb1ce970d8ee6ffcecc22351c84e67fca4/env/env.c#L194) when loading an environment. I've been reading a lot about environment and U-Boot for #19 :). `ENV_IS_IN_FAT` defaults to `y` for sunxi. Is it possible there was an environment file leftover from a vendor U-Boot?

* * *

> I might need to boot from and SD card to confirm the differences...

Same here, I gues I need to prepare an SD card. Which boot method was it using? `extlinux.conf`-compatible? UEFI? Other?

* * *

The complete log here tells me it's plausible that you had a stray environment:

```
Loading Environment from FAT... Unable to use mmc 1:2... In:    serial
```

`IS_IN_FAT` will (by default on sunxi) use `1:auto`, which AFAIUI here means internal eMMC (1), first partition marked bootable.

Thra11, 2021-06-22:
Having played with it a bit more, I think it might actually be related to _when_ it receives the key press to interrupt (So the env diff in my previous comment is probably not relevent).

I no longer have quite the same setup as when I open the issue, in that instead of a blank eMMC and a bootable SD. I now have a bootable eMMC and a bootable SD (both with Tow-boot).

Basically if it shows the extlinux prompt (with choice of nixos generation), then you press Ctrl+C, then it will drop into the two-boot menu and all of the options will fail. Let me just double check this reproduce and then see about uploading the serial output...

samueldr, 2021-06-22:
Oh! that's an important distinction.

This would happen after `run distro_bootcmd` ran. In my case, since it's running against EFI+GRUB, I can't run that *after*.

So that will help reproducing the issue. Also, that kind of confirms my initial suspicions about what I had observed, that cancelling the distro bootcmd runs with a clobbered environment, and `for` loops in hush seemingly "start back where they were" for some odd reason.

Thra11, 2021-06-22:
> Oh! that's an important distinction.

Indeed. I assume previously it was _just_ getting to distro bootcmd before I managed to interrupt it, but I didn't realise. Here are some more logs in case you want to see, though _hopefully_ you might be able to reproduce it yourself now.

```diff
26a25,26
> bootfile=/boot/extlinux/extlinux.conf
> bootfstype=ext4
29a30
> devtype=dhcp
33a35
> fdtcontroladdr=b9f2ac60
34a37,38
> fileaddr=4fc00000
> filesize=2130
38a43
> mmc_bootdev=1
48a54
> serial#=92c000ba7bd5c66f
53a60
> target_name=DHCP
61a69
> usb_pgood_delay=2000
65c73
< Environment size: 5262/131068 bytes
---
> Environment size: 5462/131068 bytes
```

[teres-boot-output-3.txt](https://github.com/Tow-Boot/Tow-Boot/files/6697106/teres-boot-output-3.txt)
[teres-boot-output-3-before.txt](https://github.com/Tow-Boot/Tow-Boot/files/6697105/teres-boot-output-3-before.txt)
[teres-boot-output-3-after.txt](https://github.com/Tow-Boot/Tow-Boot/files/6697104/teres-boot-output-3-after.txt)

Thra11, 2021-06-22:
> tested against stock U-Boot to confirm that there's something to contribute back upstream.

It never occurred to me to press Ctrl+C in the NixOS extlinux menu. Does it return you to the regular u-boot prompt?

samueldr, 2021-06-22:
Yes. Kind of... `^C` cancels the currently running command. The extlinux.conf compatible loading is just a U-Boot command. So it cancels it, since the boot fails, it continues running the `distro_bootcmd`, and eventually fails to the console.

samueldr, 2021-06-22:
Another quick thought... What if it's only *cancelled* `for` in hush that somehow restarts at the wrong location?