Tow-Boot/Tow-Boot #31: Can't get graphics with openSUSE with upstream DTB

Jojonintendo, 2021-06-13:
I've recently installed Tow-Boot, as it has been my only solution to get working graphics on my already installed openSUSE. I installed oS via serial console, to the NVMe, LUKS encrypted,

With Tow-Boot I can type my LUKS password and log into my desktop, all seems well. Except I have noticed that there are no modules loaded for sound or battery (cw2015_battery), so I tried to load another dtb from upstream. I don't know how to do it from Tow-Boot, I always get the following:
`libfdt fdt_check_header(): FDT_ERR_BADMAGIC`

So I load it from GRUB with the following:
`devicetree /boot/rk3399-pinebook-pro.dtb`

Then the output is different, for instance I now see messages related to the cw2015_battery, so the new dtb seems to be loaded fine. However I have no graphical output again. So far I've tried efifb=off, blacklist=efifb, forcing video parameters similar to how other distros do, but I can't get any video output.

I guess I have one question, as the ARM land is quite new territory for me. Where can I get the dtb that Tow-Boot loads by default? If I could compare them, it would certainly provide some interesting info.

Sorry if this isn't related to Tow-Boot, the boot process is almost black magic to me. I'd be more than happy to provide any additional logs/output. Thanks.

samueldr, 2021-06-13:
> Sorry if this isn't related to Tow-Boot, the boot process is almost black magic to me. I'd be more than happy to provide any additional logs/output. Thanks.

No worries, casting light in the darkness of ARM boot is a goal of Tow-Boot.

* * *

> I guess I have one question, as the ARM land is quite new territory for me. Where can I get the dtb that Tow-Boot loads by default? If I could compare them, it would certainly provide some interesting info.

The "dtb" that Tow-Boot uses is the internal FDT (flattened device tree), which is built as part of the U-Boot (thus Tow-Boot) build process.

In turn, this is the dtb file used to embed in the firmware:

 - https://source.denx.de/u-boot/u-boot/-/blob/v2021.04/arch/arm/dts/rk3399-pinebook-pro.dts

Note that it may (actually is) further modified by patches

 - https://github.com/Tow-Boot/Tow-Boot/blob/development/boards/pine64/pinebook-pro.nix#L23-L37

(Hopefully the patches locations are self-evident for you to look at.)

* * *

Another clue that might help you: The generic distro boot scheme, which Tow-Boot uses, after all this is an opinionated distribution of U-Boot, can load DTB files for EFI boot.

 - https://source.denx.de/u-boot/u-boot/-/blob/v2021.04/include/config_distro_bootcmd.h#L158-164
 - https://source.denx.de/u-boot/u-boot/-/blob/v2021.04/include/config_distro_bootcmd.h#L150-152

This is a bit thick to parse, so let me explain in words.

On any of the scanned boot sources for EFI, if there is a `/dtb` or `/dtb/current` directory, load the board-specific DTB files from there.

By "board-specific DTB files", it's a built-in value, something like `rockchip/rk3399-pinebook-pro.dtb` for the Pinebook Pro.

* * *

Finally, I'm not 100% positive that `efifb=off` works, even though the kernel source explicitly shows it should. I still haven't actually tried it. My system still boots using the `extlinux.conf`-compatible boot scheme. Though I do want to migrate to UEFI soon. It will need some minor work in NixOS to support properly (mainly using `/dtb/current`),

samueldr, 2021-06-13:
As a footnote, there's a behaviour I still need to properly explore and grok before really talking about it. It's related to the `chosen` node of the FDT.

 - https://source.denx.de/u-boot/u-boot/-/blob/v2021.04/arch/arm/dts/rk3399-pinebook-pro.dts#L21-23

It *may* be overriding the `console=` parameters when using the built-in FDT. Though don't assume I'm right here. It's in my TODO list, to better understand when it's used.

You may want to try running this in the firmware console:

```
=> fdt addr $fdtcontroladdr
=> fdt rm /chosen stdout-path
```

And boot without another dtb being chosen... Actually that would kind of explain the behaviour experience by another user I guess.

Jojonintendo, 2021-06-13:
Many thanks for your detailed replies, I appreciate it. Now I have more things to test!

The dtb-rockchip package provided by openSUSE has the following structure:

```
/boot/dtb
├── rk3399-pinebook-pro.dtb -> rockchip/rk3399-pinebook-pro.dtb
└── rockchip
    |── rk3399-pinebook-pro.dtb
```

Tow-Boot should therefore already be loading the correct file, however if I force this same file via GRUB the behaviour is different, as per my first post. In fact I remember this package not being installed by default in openSUSE, I added it later but it doesn't seem to have any effect by itself.

I'll keep digging and investigating, willl report if I find anything useful. Thanks again for Tow-Boot! It's already so much better for my Pinebook Pro.

samueldr, 2021-06-13:
> After these commands, I open the menu again and select the NVMe (where / is), but it isn't recognized and can't boot.

Oh, that's another issue relatively unrelated here... Let's ignore that for now.

* * *

> Tow-Boot should therefore already be loading the correct file, however if I force this same file via GRUB the behaviour is different

Odd. `/boot` is the ESP, right?

Jojonintendo, 2021-06-13:
> > Tow-Boot should therefore already be loading the correct file, however if I force this same file via GRUB the behaviour is different
> 
> Odd. `/boot` is the ESP, right?

Sorry, I should have paid more attention to your previous message. I'm using the default layout from openSUSE's installer, which makes /boot/efi the ESP. The rest is under btrfs subvolumes. This is the structure:
```
/boot/efi/
├── EFI
│   ├── boot
│   │   ├── MokManager.efi
│   │   ├── bootaa64.efi
│   │   ├── grub.cfg
│   │   └── grub.efi
│   └── opensuse
│       ├── grub.cfg
│       └── grubaa64.efi
└── ubootefi.var
```
Tomorrow I'll try to put a symlink to dtb in ESP and report back.

samueldr, 2021-06-13:
The symlink won't work as expected across filesystems.

You will have to copy the whole `dtb` folder (really, the pinebook-pro DTB) for this to work.

Jojonintendo, 2021-06-14:
You were right, as expected with the DTB in the ESP, I no longer have the error message `libfdt fdt_check_header(): FDT_ERR_BADMAGIC`, and the file loaded correctly on boot. I'll try to use another DTB, from other distros maybe, and also compare them to understand where the issue could be.

Jojonintendo, 2021-06-15:
After some experiments I've found that Manjaro's DTB doesn't provide graphics, but Fedora's does. As a bonus, I also have a battery gauge, which wasn't present on U-Boot's upstream DTB. There are probably some other things missing, like sound (Pulseaudio only has a dummy output, aplay -l doesn't find a single device), but for now graphics are solved.

Now I understand this statement on the front page and appreciate your work even more: 

> Should all the distributions have to build and manage all the BIOS for all your boring x86_64 machines too?

The quest to find the most compatible DTB doesn't look really interesting, but I'll do my best and report my findings upstream. Thanks again for your help.

closed 2021-06-15 by Jojonintendo

Jojonintendo, 2022-01-26:
@13werwolf13 Unfortunately, I no longer have a Pinebook Pro, so I can't be of much help here. I know some changes were made last year to openSUSE's dtb for rockchip, but I don't know exactly.

13werwolf13, 2022-01-26:
@Jojonintendo sorry to hear that. I hope you just lost pbp of your own free will and not for tragic reasons.

Jojonintendo, 2022-01-26:
@13werwolf13 Oh no it worked fine, I just sold it because I was getting tired of maintaining it that much instead of actually using it :p