Tow-Boot/Tow-Boot #37: Boot Loader Specification support?

flokli, 2021-06-25:
Does Tow-Boot support [BOOT_LOADER_SPECIFICATION](https://systemd.io/BOOT_LOADER_SPECIFICATION/)?

This provides an already common standard out there, which should be sufficient enough even for embedded stuff.

samueldr, 2021-06-25:
Not yet. But maybe it shouldn't. Instead it would be better advised to run systemd-boot IMO.

In fact, in theory it should already be able to do it, but when I tried the NixOS tooling (or systemd's) failed to install.

And Tow-Boot, as a whole, targets "generic boot" scenarios. It's less intended for "embedded" uses, where the boot chain needs to be more strict and controlled by the project.

Not sure this adequately answers the underlying concerns.

flokli, 2021-06-26:
> Not yet. But maybe it shouldn't. Instead it would be better advised to run systemd-boot IMO.

I disagree. This is meant to be a specification implementable by multiple bootloaders, and writable by multiple distros/operating systems - so a generic standard, that Tow-Boot could just start supporting - this seems way cleaner than supporting yet another format, such as "`extlinux.conf` dd'ed to a block device".

Also, `systemd-boot` isn't supported on many embedded systems - only those with UEFI support, while u-boot is.

samueldr, 2021-06-26:
I guess I was right that, maybe from a lack of proper documentation on my part, you wouldn't see it as I do.

(I will answer in an order that better fits the narrative.)

> I disagree. This is meant to be a specification implementable by multiple bootloaders

This is right. But Tow-Boot is not "a bootloader"... Maybe I should have named it "Tow-Firm"?

Just like U-Boot, it does much more important things, but it is named after the feature that is the "least important" to me...

> Also, systemd-boot isn't supported on many embedded systems - only those with UEFI support, while u-boot is.

... and that is because any system supported by U-Boot, and thus Tow-Boot supports UEFI. This fact is the reason I don't think it is important that the firmware itself supports the protocol. To me Tow-Boot is "your BIOS", for your boring old computer. The fact that it knows how to boot a Linux kernel is an inconvenient truth.

So really, what Tow-Boot currently supports is *whatever the default distroboot boot chain from U-Boot* is.

Though I'm not saying there's no value in adding support for the boot loader specification. There would be. It would also have value upstream. What I'm saying is that it's not where my effort stands right now.

*My* main goals here are two fold:

  - Make booting to "a standard firmware interface" and "a standards-based OS" less confusing across the different platforms.
  - Work on the user interface of "a standard firmware interface".

For the former, it's already mostly done. Documentation is superficial, but present. You either install to the dedicated storage (SPI Flash) when present, or use the EBBR-compliant "shared storage" strategy, as a base image. That base image **does not contain an operating system**. It only contains enough to boot a *standard* image. The expectation is that the user boots a UEFI-based system from USB. Though since we're building upon the building blocks from U-Boot's generic distroboot, it continues supporting those other schemes too.

As for the user interface, some early work has already been done. Rather than dropping the user to a user-unfriendly shell for commands, "stopping" the boot process provides the user with a menu interface to select a boot source. Future goals include building a more in-depth interface to provide the "usual" options, like editing the boot order and saving it to the firmware's environment, or changing some hardware options (e.g. overclocking, changing some DT node values, changing config.txt nonsense on Raspberry Pi systems).

Anything else falls in the "sounds nice, but **I** don't have the time to dedicate for that".

* * *

> this seems way cleaner than supporting yet another format, such as "extlinux.conf [...]".

I personally think supporting the UEFI iso on a USB drive is the cleanest and more standard alternative. Supporting the boot loader specification gives us compatibility where that exists. For now it generally requires a boot loader of some sort to know about that specification.

While the same is just about true for UEFI, the bigger difference is that UEFI *already is* deployed in different use cases on ARM platforms. So supporting UEFI boot as first-class citizen gives us better compatibility in the existing ecosystem. Any efforts producing an OS for UEFI works on those platforms too (given the same condition that e.g. mainline linux works on those, nothing different).

> such as "extlinux.conf dd'ed to a block device".

Maybe I'm missing a detail here. Whats the difference between dd'ing an extlinux.conf-based image to block device, dd'ing a boot loader specification-based image to a block device or dd'ing a UEFI boot-based image to a block device?

Or rather what would a boot loader specification system provide that makes it different here?

* * *

> Also, `systemd-boot` isn't supported on many embedded systems - only those with UEFI support, while u-boot is.

Right. I don't intend this *opinionated distribution of U-Boot* to replace U-Boot for all use cases! It's a superset building on top of U-Boot. For embedded uses I will continue suggesting working from U-Boot, or maybe using Tow-Boot as a base to build upon. Tow-Boot is a **strongly opinionated** firmware that aims to be "boring like your laptop's BIOS/UEFI (should be)".

Most of what I described here is already possible with U-Boot. The main difference with U-Boot *as it is currently* is that Tow-Boot has a tooling to build distribution-independent binaries for end-users.

I expect that in an embedded use case, the embedded product implementer will produce "freestanding" images, with everything already built-in. After all you don't want to go around installing an operating system after the fact on all those gizmos you produce :).

samueldr, 2021-06-26:
And to paraphrase more succinctly: "Yes, BLS would be good in U-Boot. I *personally* won't work on it in the foreseeable future"

samueldr, 2021-06-26:
I've just taken a look at the Boot Loader Specification. And there's one huge deficiency (shared with many other bootloadery bits). It assumes the device tree binary to be loaded is a static value, and not derived from other knowledge.

This means that with a BLS-following installation you have two choices:

 - Load a specific device tree binary from an hardcoded path.
 - Not load a device tree binary and hope the previous steps provide a good enough one for that generation.

This is probably the main advantage the non-standard `extlinux.conf`-compatible scheme has within U-Boot; it uses a firmware-provided dtb filename. And probably the most inconvenient truth about booting on ARM. Providing the "right" DTB for the kernel is a total mess (and another topic we shouldn't discuss in this particular issue).

samueldr, 2022-11-08:
With 2022.07, some yet-unexplored-by-me support for BLS has been added

 - https://source.denx.de/u-boot/u-boot/-/blob/v2022.07/doc/develop/bootstd.rst

Read the latest revision of the docs (might not apply to 2022.07) here

 - https://u-boot.readthedocs.io/en/latest/develop/bootstd.html#u-boot-standard-boot