Tow-Boot/Tow-Boot #45: "Offset exceeds device limit" while updating pinebookPro SPI to 002

badcold, 2021-06-29:
Hello. I've been using Tow-Boot "pine64-pinebookPro-2021.04-001" with some success (and some issues but likely unrelated to Tow-Boot).
I'm trying to update to 002, as described in the release page it fails to recognize `$board_identifier`. The suggested workaround, `env default board_identifier`, didn't work ($board_identifier stays empty). 
I had to do `env set board_identifier pine64-pinebookPro` to pass the check.

Then this happens:

[image]

Any idea?
I would like to avoid having to completely erase the SPI Flash as it's been annoying to force booting on the SD without Tow-Boot.

samueldr, 2021-06-29:
Ooooh...

I guess I never tried upgrading from 001 stock to 002.

See the `Unknown command 'setexpr'`? It's a missing command that the script now relies on.

You should be able to run these commands in the Firmware Console. Note that these will not "harden" against partial flashes like the updater script does. It's not likely to be an issue unless you decide to force reboot the Pinebook Pro while it's writing.

**I have not tried running these commands yet.** I'll try in a few hours once I'm done with work. I would suggest you wait until I verified they'll work before trying :).

```
mmc dev 1
load mmc 1:2 $kernel_addr_r Tow-Boot.spi.bin
sf probe
sf update $kernel_addr_r 0x0 $filesize
```

samueldr, 2021-06-29:
*Added to the known issues*. This message is for other users, @badcold, I know you won't be able to erase the SPI and boot from the new release.

* * *

## Workarounds

If it is not a problem for your situation, erase the SPI Flash content, and reboot into the firmware from the SD card.

> *How do I know if this is a problem for me or not?* Generally, it will be a problem only if you have a valid U-Boot install on the eMMC, which would be preferred over the SD card.

Otherwise, \[wait until the previously listed commands are tested and\] use the previously listed commands to manually upgrade the firmware in SPI Flash.

samueldr, 2021-06-30:
@badcold, here's my transcript of going from the `001` release (without a change) to `002` on another RK3399 device. Note that I burned `spi.installer.img` to an SD card.

```
[Selected Firmware Console in the menu]

Use `run menucmd` to return to the menu.

=> mmc dev 1
switch to partitions #0, OK
mmc1 is current device
=> load mmc 1:2 $kernel_addr_r Tow-Boot.spi.bin
1566248 bytes read in 72 ms (20.7 MiB/s)
=> sf probe
SF: Detected w25q128 with page size 256 Bytes, erase size 4 KiB, total 16 MiB
=> sf update $kernel_addr_r 0x0 $filesize
device 0 offset 0x0, size 0x17e628
1246760 bytes written, 319488 bytes skipped in 11.471s, speed 139816 B/s
=> 
```

After, I removed the SD card (to ensure things worked), and used `reset` to restart the machine.

Note that if you're using serial to access the machine, the baud rate was changed. I'm saying so since I scared myself into thinking the machine didn't boot at first :).

samueldr, 2022-02-09:
*Closing since upgrades from 001 to 002 are now unlikely, the documented steps are still valid.*

closed 2022-02-09 by samueldr