Tow-Boot/Tow-Boot #49: Unrecognised filesystem Type

UltraBlackLinux, 2021-07-01:
Hey, there,
I am currently trying to boot from my usb thumbdrive and sdcard - And it just won't work; It always sais "Unrecognised filesystem type"
I have already tried either both inbuilt usb ports and my sdcard slot on the Pinebook Pro.
I have flashed clonezilla onto both devices via Etcher.
Trying yet another different device, this is what I got this time:
```
** Unrecognised filesystem type **
Scanning disk sdhci@foo.blk
Scanning disk usb_mass_storage.lun0
Found 7 disks
** Unable to read file ubootefi.var **
Failed to load EFI variables
BootOrder not defined
EFI boot manager: Cannot load any image
USB Boot failed.
Press any key to continue...
```
After another usb rescan, I got this while trying to boot my usb device:

```
Device 0: Foo
... is now current device
libfdt fdt_check_header(): FDT_ERR_BADMAGIC
BootOrder not defined
EFI boot manager: Cannot load any image
USB Boot failed.
Press any key to continue...
```

While creating this issue, Tow-Boot just decided to freeze up - seems to happen, when it's not touched in a while (5-10min).

Little sidenote: Please set up github actions - At the moment I can't compile the package myself.

I hope this info is useful!
Thanks!

samueldr, 2021-07-01:
Discussion on the matrix channel showed that an *x86_64* image for clonezilla was used. Booting on an AArch64 CPU requires AArch64 operating systems (32 bit ARM could work in theory, but don't *expect* them to work *in practice*).

The exact error message listed *could* be a red herring from one of the multiple steps used by the generic distro boot mechanism.

UltraBlackLinux, 2021-07-10:
fixed in a long discussion: my sdcard was the cause; workung, but the pbp didn't accept it

closed 2021-07-10 by UltraBlackLinux