Tow-Boot/Tow-Boot #52: Shared Storage guidance (fdisk warns about GPT?)

colemickens, 2021-08-28:
Hi.

Maybe this is because my SD came pre-formatted, but after I `dd` the rpi4 `shared.disk-image.img` to my SD Card... I figured I'd fire up `fdisk` to add a traditional `/boot` and `/root` for a shared media install.

However, fdisk gave me this warning, and I'm not actually sure what I want to do here:

```
/home/cole/downloads/raspberryPi-aarch64-2021.04-002> sudo dd if=shared.disk-image.img of=/dev/sdb bs=4M; sudo sync; sync
8+1 records in
8+1 records out
34620416 bytes (35 MB, 33 MiB) copied, 0.018304 s, 1.9 GB/s
```

```
/home/cole/downloads/raspberryPi-aarch64-2021.04-002> sudo fdisk /dev/sdb

Welcome to fdisk (util-linux 2.36.2).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

The backup GPT table is not on the end of the device. This problem will be corrected by write.
A hybrid GPT was detected. You have to sync the hybrid MBR manually (expert command 'M').
```

Any tips? Should I ignore it? Should I have "wiped" the sd card more thoroughly?

samueldr, 2021-08-28:
> The backup GPT table is not on the end of the device

That's expected, as the image wasn't the exact size of your storage medium.

Not sure about the second (unrelated AFAICT) message about the hybrid MBR; I find fdisk quite naggy and needlessly hard to use personally. Though it sounds like it means "if you edit the GPT the MBR data will not be synchronized", which is fine as long as you don't edit the size or location of the Tow-Boot partition.

flokli, 2022-04-16:
So is this a matter of "documenting how to get from flashing the shared storage to installing the rest of the OS"?

 - resizing the partition to the physical size
 - making partitions for the OS you want to install (and expectations w.r.t. extlinux-compatible bootloader config)

closed 2022-06-13 by samueldr