Tow-Boot/Tow-Boot #59: Raspberry Pi Zero 2 support

colemickens, 2021-11-06:
Just opening this for a place to track, share notes, etc.

samueldr, 2021-11-06:
Note that when discussin with @leo60228, I realized there is a [generic `rpi_arm64` U-Boot defconfig](https://source.denx.de/u-boot/u-boot/-/blob/master/configs/rpi_arm64_defconfig), which may be better to use, rather than smushing together two U-boot binaries in the platform firmware partition.

leo60228, 2021-11-07:
From some testing, that defconfig requires the Broadcom firmware to provide a mainline-flavored DT. This can be done with `upstream_kernel=1` in config.txt, which will look for a DTB with a mainline filename and fall back to using `overlays/upstream.dtbo` with a vendor-flavored DT (presumably `overlays/upstream-pi4.dtbo` is used on Pi 4, but isn't mentioned in docs).

boosterl, 2022-03-23:
With the [current release](https://github.com/Tow-Boot/Tow-Boot/releases/tag/release-2021.10-004) it is already possible to boot Tow-Boot on the 02W by adding a conditional filter to the `config.txt` file. Adding a conditional filter for the pi02 which selects the `Tow-Boot.noenv.rpi3.bin` kernel makes the system bootable. More information about the conditional filters can be found [here](https://www.raspberrypi.com/documentation/computers/config_txt.html#conditional-filters). The config file looks like this after the change;
```
[pi02]
kernel=Tow-Boot.noenv.rpi3.bin

[pi3]
kernel=Tow-Boot.noenv.rpi3.bin

[pi4]
kernel=Tow-Boot.noenv.rpi4.bin
enable_gic=1
armstub=armstub8-gic.bin
disable_overscan=1

[all]
arm_64bit=1
enable_uart=1
avoid_warnings=1
```
I also added the pi02W dtb file (`bcm2710-rpi-zero-2-w.dtb`), I do not know whether this is actually necessary.

leo60228, 2022-03-23:
I'd be concerned about relying on that, since it's embedding a Pi 3 device tree.

boosterl, 2022-03-23:
That's right, but I presumed it would work because I read [here](https://www.raspberrypi.com/news/new-raspberry-pi-zero-2-w-2/):

> Priced at $15, [Raspberry Pi Zero 2 W](https://www.raspberrypi.com/products/raspberry-pi-zero-2-w/) uses the same Broadcom BCM2710A1 SoC die as the launch version of Raspberry Pi 3, with Arm cores slightly down-clocked to 1GHz, bundled into a single space-saving package alongside 512MB of LPDDR2 SDRAM.

While I wait for the unified kernel, this seemed like a good temporary solution to make it work. I'm definitely not relying on this, just using this for testing purposes. I just thought this maybe could be of help for someone. But I should have added that this hack should definitely not be used for production workloads ;)