Tow-Boot/Tow-Boot #60: Post-dd instructions (for raspberryPi at least)

colemickens, 2021-11-06:
Again, leaving this as a note to self, I'll try to send this as documentation or something:

1. Use `dd` to copy tow-boot on the front of an SD card.
2. Use `parted` then `print` to get a prompt to expand GPT to the full disk
3. Then you can use `gdisk` to add a new "linux filesystem" partition behind the tow-boot partition
4. Format that ext4 partition

Then install your favorite way.

It'd be nice to have specific commands to execute to do step 2, but I couldn't figure it out quickly.

samueldr, 2021-11-07:
Note that the Raspberry Pi family of devices *may* drop GPT builds and instead use strictly MBR due to some partitioning tools behaving badly.

colemickens, 2021-11-07:
Hm ok. I see the "may", but that seems like an odd trade-off, and I say that as someone who lost a chunk to `fdisk` indeed behaving a bit counter to my expectations.

samueldr, 2021-11-07:
I wanted to look at the problem again before committing to using only MBR. It's more likely to switch to MBR than keeping GPT+hybrid MBR.

piegamesde, 2023-06-20:
It looks like in the meantime indeed a switch to MBR-only was made, and now I'm a bit at a loss on how to proceed after flashing the `shared.disk-image.img`. The OS installation instructions link to the wiki which links to the manual, which tells me to do GPT-specific things when working with UEFI. (And if I'm not mistaken, the ESP cannot even be on an MBR disk.) What is the correct way to proceed then?