Tow-Boot/Tow-Boot #64: Pinebook Pro: 'Boot to SD card' option (second option) not working

cobratbq, 2021-12-02:
I have just flashed the '002' pre-release to a Pinebook Pro SPI flash. Happy user. Thanks!
Booting works. _Except for the boot menu option "Boot to SD" does not actually boot from an SD card._

__NOTE__ I can put the same SD card in a USB card-reader connected to USB, and "Boot to USB" will work correctly. (I have previously used the same SD card to install an OS, so it should be working. I originally flashed Tow-Boot to SPI flash from the same slot.)

samueldr, 2021-12-02:
This is odd, I've used `-002` to boot from SD card successfully.

Have you tried the same operation multiple times, from power off?

Was attempting an SD card boot the first operation from power on?

Another thing to look at, but not sure how much help it will be, is to run `run bootcmd_mmc1` in the firmware cosole.

I need to look into it, but I've observed in non-reproducible manners that U-Boot's shell scripting has some sort of *weirdness* when re-running loops. It might be that it *somehow* is only trying the last boot method in some limited situations.

cobratbq, 2021-12-04:
You are correct. I performed 2 boots from power-off -- boot from USB card reader and boot from SD-card slot -- and both worked as expected. So this is a false report. Thanks for your quick response!
Also, I hope you will continue with these images, such as the SPI flash image. This has worked very well for me.

closed 2021-12-04 by cobratbq

samueldr, 2021-12-04:
Hm, annoying that there was this one (or maybe more than one) issues booting, but can't be reproduced anymore. Ideally I'd like issues to be reproducible! :)

And you shouldn't worry, I fully intend to continue improving Tow-Boot. Though this is not the main project I'm maintaining, this is why development is a bit *slow*.

To be fair, since it's been *pretty solidly stable* I don't have a lot of reasons to spend lots of time working on this :).