Tow-Boot/Tow-Boot #66: Add ARM Trusted Framework for Rpi4 to enable kexec?

colemickens, 2021-12-16:
I can't really get `kexec` to work on Raspberry Pi 4, even with upstream kernel. This article suggests that the ARM Trusted Framework might assist without requiring the user to limit their cpu for their initial boot.

https://www.thegoodpenguin.co.uk/blog/booting-linux-from-linux-with-kexec/

I don't know if this is true, though.

samueldr, 2021-12-16:
Interesting, it would need to be investigated if it applies to the Pi family of devices.

I was led to believe "TrustZone" stuff was handled already by the proprietary firmware. Anyway, it must be since there's no TF-A (ARM Trusted Firmware) needed.

Is `kexec` supposed to work with an entirely-vendor-based system?

colemickens, 2022-01-10:
I have ARM Trusted Firmware working for RPi4 in Tow-Boot. I'm not sure if I'll send a PR for it though, given that I can't seem to tell that it matters much (kexec gets further, but still doesn't work - there might be some USB teardown necessary - not going to look into that though).

Also, a reason to not pursue --> it seems to **necessitate** blocking extlinux from advertising the upstream DTB so that the one passed to u-boot from firmware is used. If this isn't done, the upstream dtb will be used, and then only a single core will come up (and likely the Watchdog will reboot you, if enabled, for booting too slowly).

samueldr, 2022-01-10:
Keywords I don't actually know enough about that may be relevant:

The **PSCI** is AFAIUI involved for setting up cores. I don't know what it actually means in this context at this point in time.

colemickens, 2022-01-10:
Yes, I was able to confirm that I saw messages about PSCI in kernel log and saw some associated devices or sys entries or something related to it. And when I mentioned that 'kexec' got further - I meant that it didn't complain about being unable to park the other CPU/cores (indicating PSCI available+working), and went all the way into shutting down. (But ultimately it fails to complete or boot, with some "sounds related to kexec" usb messages being the last thing printed out. This was actually the same behavior I observed when booting without-ATF, with-`nr_cpus=1`, which should've also allowed to `kexec` in theory)

I opened a related issue, too, which I mostly link to show the exact thing I tried with kexec: https://github.com/raspberrypi/firmware/issues/1675#issuecomment-1008696856

thesaadmemon, 2023-02-22:
Same issue here, do we have any updates on this one, or if it works with any other models?