Tow-Boot/Tow-Boot #68: [Pinebook Pro] Cold NVMe Boot Doesn't Work Reliably

delipunch, 2022-01-07:
My PBP has no eMMC or SD Card in it, it has Tow-Boot in the SPI flash, and Manjaro-ARM on an NVMe drive.

Cold booting the machine usually doesn't work, resulting in the boot menu appearing. From here, selecting `reboot` will normally allow the machine to successfully boot afterwards.

It's as if the NVMe drive does not have enough time to initialize the controller properly from a cold boot. Maybe an option to have a delay before attempting to boot from a device will fix this?

Tow-Boot version used: https://github.com/Tow-Boot/Tow-Boot/tree/release-2021.04-002
NVMe Drive used: `KINGSTON RBUSNS8154P3256GJ1`, as reported by `nvme list-subsys`

samueldr, 2022-01-07:
Interesting you've noticed it works after a reset.

The most I knew about the situation (I don't have an NVMe drive to test with in the PBP), it "didn't work correctly" (unqualified statement) with U-Boot when installed to SPI.

The good news here is it's AFAIK not specific to Tow-Boot.

* * *

Some things to look at, to help understand the situation:

Does the `nvme list-subsys` command work on a cold startup of the machine? Or is it only working under the same conditions required for NVMe to boot to the OS?

*If* `nvme list-subsys` works with a cold boot, does breaking into a console, and then going back into the menu to select NVMe boot work?

delipunch, 2022-01-10:
I've since replaced the 256GB Kingston drive with a 500GB Crucial P2 (`CT500P2SSD8`) and cold boots seem to be more reliable, so I've not had a chance to try the `nvme` command in the boot console upon a failed boot.
If it fails to boot into the OS again with this new drive, I'll try to see how the `nvme` command reports things. It might just be a case of drive-specific quirks.

delipunch, 2022-01-20:
For what it's worth, the rating for Power State 0 of the Crucial drive is 3.5W max, compared to ~8W for the Kingston's Power State 0.

Additionally, the Crucial P2 hasn't failed to cold boot on me yet so this issue could be closed, but I'd prefer to have more data points to ensure that the drive initialization timeout period for Tow-Boot isn't problematic for a majority of NVMe Drives. However If you feel that this issue _should_ be closed, I won't argue otherwise.

samueldr, 2022-01-20:
I don't think it should be closed, I would instead be interested in seeing experience with misc. other rk3399 boards, and SSDs!

It's still true it's not reliable *for all drives*, and now it needs to be known *why*.

samueldr, 2022-02-09:
Note that *needs information* here is from other people, and from other configuration. @delipunch all you provided was plenty.

Strit, 2022-07-23:
Just tried an [Intel 660p 512 GB](https://www.intel.com/content/www/us/en/products/sku/149403/intel-ssd-660p-series-2-0tb-m-2-80mm-pcie-3-0-x4-3d2-qlc/specifications.html) and Tow-Boot 2021.10-004 on my Pinebook Pro and it does not detect it at all.

remexre, 2022-09-14:
Also having problems with an Intel 660p, PCI ID 8086:f1a8. My symptoms match https://forum.manjaro.org/t/nvme-suddenly-stopped-working/4507 pretty much exactly -- https://github.com/pcm720/u-boot-build-scripts/blob/fb2b1f3ddeb552f51379a57118ccc4109a5854a8/patches/u-boot/0007-address-NVMe-instability.patch sounds like the relevant patch?

samueldr, 2022-09-14:
This patch should still apply

 - https://source.denx.de/u-boot/u-boot/-/blob/v2022.07/drivers/nvme/nvme.c#L30-46

It may be fine to enable outright, I guess. Another thing to explore is looking around if there were upstreamed changes related to NVMe timeout on U-Boot's mailing lists.

remexre, 2022-09-14:
Okay, I'm pretty confused now. Assuming I didn't do anything wrong in https://github.com/remexre/Tow-Boot/commit/c1e60b2943cbbb5ee690ff30f79d99ed10318fbb , it didn't fix my issue. However, I mounted it in another machine (an x86 PC) and it worked there... then upon reinstalling it into the device, it started working again.

Maybe there's some power-saving-related register that gets saved in the device, and got reset by Linux, but prevents being properly initialized by Tow-Boot?

samueldr, 2022-09-14:
Unclear. It might be that doubling it isn't enough but *just* on the threshold, and you got lucky. Try rebooting *and* full poweroff a bunch of times and see if it stays reliable.

If it is not, doubling again is probably safe. I assume it'll just take longer to fail to start the NVMe subsystem.

tdcook, 2022-09-18:
Also using an Intel 660p, I'm seeing the same behavior as @remexre where NVMe boot will work one time after removing and installing the drive in another machine then re-installing, but then fail on subsequent boots. Leaving the Pinebook powered off for some time (i.e. several days) also exhibits this behavior. Increasing the timeout multiplier doesn't seem to work. I suspect this is a different issue than the one the timeout increase is supposed to resolve.