Tow-Boot/Tow-Boot #69: rasperberrypi-eeprom update instructions

colemickens, 2022-01-10:
I updated my `raspberrypi-eeprom` package that I use with the nixpkgs that I build my system with, and updated the system.

But for the rpi to update its own eeprom "in-place" (without a separate SD card to host a `recovery.bin`), there's a bit more to it:

```
sudo mount /dev/{your_tow_boot_fw_part} /boot/firmware
sudo env FIRMWARE_RELEASE_STATUS=stable BOOTFS=/boot/firmware rpi-eeprom-update -a
sudo reboot
```

Also, it seems rpi-eeprom-update wants to use the "default" release which lags behind "stable" by quite a bit. Don't ask me.

samueldr, 2022-01-10:
We probably should consider making a purpose-built Linux system that serves to run the raspberry pi updater...

I have some code using my WIP non-NixOS "appliance" tooling that might be useful for that (and other features on other boards). Main reason it might be useful is it produces *extremely minimal* images, e.g. it would likely be a 4MiB file.

samueldr, 2022-02-09:
> **cleverca22** 18:27
> if you put `recovery.bin` + `pieeprom.upd` in the fat32 of your disk image
> then it will re-flash the eeprom on first boot
> clearing anything the user had
> it will even boot with an erased eeprom
>
> when using pieeprom.upd, it will rename recovery.bin -> recovery.000, and then reboot
> 
> causing it to boot normally on the 2nd time around

I guess this answers the question about how we could handle shipping updates to the eeprom **and** having a known eeprom firmware build in use.