Tow-Boot/Tow-Boot #71: automated testing

colemickens, 2022-01-12:
Ahead of my nixpkgs changes merging, I thought I'd try my hand at a script to compare against a "LKG" of the platform/board binary.

Some quick thoughts/questions:
1. Should there be a slight refactor so that something exposes a combination of `{ image = ...; tow-boot-binary = ....; }` so that it's easy to walk to output boards and have the corresponding platform binary inside that we want to test?
2. Alternatively, it seems like some of the boards are basically just wraps around the builder, indicating maybe that we don't want to have a binary for each board necessarily? (This even applies to rpi4 where I have two board outputs, one with and one without ATF, where there's only a single TB binary to test).

Just stashing some thoughts, curious what you think or have a preference or if I'm overthinking it.

samueldr, 2022-01-12:
As for *slight refactors* and such, I expect to make a big refactor and the builder for Tow-Boot will be totally unrecognizable. The main goal is going to make it much easier to configure, and more resilient. As an example I will integrate the configuration validation harness that is now in Mobile NixOS's kernel builder so that we can be more confident that additional configuration options are enabled.

So anything is up for consideration.

* * *

I don't follow why you need the binary here. Can you explain? The current expected update path will be to `dd` the partitions, including for the Raspberry Pi. Editing the filesystem through file copy/move operations may cause more hurt down the line. Though just now I'm thinking that for the dastardly Raspberry Pi family it might not be correct as `config.txt` would be reset...

colemickens, 2022-01-20:
Roughly the thought was to restructure outputs so that it's easy to access:
* board.image
* board.firmwareBinary
* (maybe even) board.updateScript to update the eeprom/firmware/etc files

Access to the firmware binary was to be able to script capturing and then checking against "known good" firmware binary hashes. This would allow potentially even automating nixpkgs updates in the majority of cases where the u-boot binary is not changing. (something we discussed a couple weeks back)