Tow-Boot/Tow-Boot #72: "vendor" in rpi firmware?

colemickens, 2022-01-12:
I noticed that it seems like you copied the ARM TF into this repo. As I was typing up #71, I was going to point out that automating nixpkgs updates means that I automate rpi4 firmware flowing into here eventually (which also notably wouldn't be covered by #71).

Presenting the question, does it make sense to copy the rpi firmware into the overlay (or otherwise package it in this repo) to control the version being deployed?

(If not, does it make sense to remove ARM TF or upstream whatever you did to it back to nixpkgs?)

For example, we write the config in this repo, but are dependent on the firmware from nixpkgs. Granted, I think this config format is basically stable, but just something for consideration.

samueldr, 2022-01-12:
I actually don't know yet. I still need to think whether I want to depend on Nixpkgs *only for the build tooling* and always rely on local derivations for artifacts, or if it should rely on Nixpkgs for build artifacts.

Though I have been thinking it should handle its requirements as we will most likely end up customizing more and more parts of the build. Furthermore, when I move to the modules system for describing the builds, I most likely will not use the "builder" as it is made currently, and instead rely on the modules system for TF-A and all similar artifacts.

So I guess the vendor firmware bits should all be declared locally.

colemickens, 2022-01-28:
>So I guess the vendor firmware bits should all be declared locally.

Oh lord I just went through the hassle of making it auto-updateable in nixpkgs, but I guess ultimately that should lift and shift here.

At the very least, it seems odd that we source u-boot from nixpkgs, but maintain a separate patch set here.

samueldr, 2022-01-28:
It really could go either way, but keeping things like they are right now means we couple tightly the vendor binaries lifecycle with our Nixpkgs update cycle.

Though luckily, it should be trivial enough to port most of what you did into our overlay.

colemickens, 2022-01-28:
Okay. Presumptions:

1. we'd copy from nixpkgs, rather than move (not sure we really could anyway)
2. we'd only copy over the u-boot binary, patches, and the foundation's firmware, so that tow-boot is less dependent on nixpkgs's versions and patches aligning. (Though, I guess in theory we're also going to copy foundation+mainline dtbs in, but I'm not sure mirroring the kernels in here make sense, oh look it's this "fun" again.)

I'll take a crack at "copying" the firmware part of my update into tow-boot's overlay.

colemickens, 2022-01-28:
Hm, how do you feel about Tow-Boot packaging / prescribing an rpi-eeprom version? @samueldr

samueldr, 2022-01-28:
It most likely will be required, to make sure things work as expected. I actually had a small thought about it in the past few days, and was wondering if it was possible to *force* it. Additionally adding the required tooling in U-Boot to get information like the version, and maybe even configure/flash it.

samueldr, 2022-02-09:
(See #69 for details about forcing the rpi eeprom)