Tow-Boot/Tow-Boot #75: rpi4: upstream_kernel=1 yet we copy in the foundation/vendor DTBs

colemickens, 2022-01-20:
Another angle of the usual weird problem with the RPi.

Right now things work for me, but I think it's a bit of luck. Also, I notice that when I boot the mainline kernel, I get slightly different device names and consoles depending on which bootloader is used. I think this is because extlinux is loading mainline kernel and re-loading the mainline DTB as the device tree. On the other hand, since NixOS's grub generator currently omits DTB lines, grub loads mainline with the firmware-mangled, vendor-sourced, foundation-kernel-derived DTBs.

It almost feels like the right thing is to have multiple variants... but I'm not sure.

Selfishly, I'd like to replace the vendor DTBs with (any recent version) of mainlines, but I don't know if one direction of version skew is more forgiving than another.

samueldr, 2022-01-20:
One of the goals I want to work towards, I think you know, is a firmware interface, e.g. a "BIOS menu".

Part of the options I always assumed would be needed was configuring *most* `config.txt` options. Additionally, I expect to allow selecting between "vendor flavoured" and "mainline flavoured" default provided FDT.

*In addition to all that*, I need to figure out a way, probably involving the environment, to import and save an arbitrary dtb file to provide to the OS, for all boards. Still unsure whether it should be only a path being saved, or the whole blob.

I *think* that would solve the conundrum, since for the Raspberry Pi, we'd **always** at least load a bundled FDT, unrelated to the one the vendor firmware prepares for U-Boot.

colemickens, 2022-01-20:
Yeah, for this exact case, just allowing "upstream_kernel" to be preboot-time modifiable should be enough, both the upstream +  vendor DTBs could be available in different directories, per that flag.

I hadn't even considered other boards - I guess I'd assumed if they weren't in a similar situation of having firmware loaded+altered DTBs that we would just let the NixOS bootloader generation handle the configuration of DTB application. But after reading your comment, I realize now that the (roughly over-simiplified) goal of "portable /boot" implies that (tow-boot) is getting the DT ready.

:/

samueldr, 2022-01-20:
> I hadn't even considered other boards - I guess I'd assumed if they weren't in a similar situation of having firmware loaded+altered DTBs that we would just let the NixOS bootloader generation [...]

They aren't in the exact same situation, but if you're using a boot method that is not aware that it needs to load the board-specific DT for the kernel, you might want to choose a specific device tree. Or that specific device tree may hold user customizations!

Similarly, customizations *may* be supported through overlays, or overlay-like semantics. Thinking about RK3399, like the Pinebook Pro, it would be nice if as an end-user you could *just* enable some options in the BIOS to enable basic underclocking or overclocking.

mupuf, 2022-09-17:
> Part of the options I always assumed would be needed was configuring most config.txt options. Additionally, I expect to allow selecting between "vendor flavoured" and "mainline flavoured" default provided FDT.

Is this the reason why the raspberryPi-aarch64/default.nix is generating multiple targets (`rpi_3_defconfig` and `rpi_4_defconfig`) rather than relying on `rpi_arm64_defconfig` which would work on more than just the Raspberry Pi 3b and the Raspberry Pi 4b?

In my case, i have the Raspberry Pi 3b+ and the Raspberry Pi CM 4 which both fail to work with tow-boot by default (RPi3 does not have ethernet nor keyboard support, while the CM4 doesn't boot without updating start4.elf and also does not have ethernet nor USB). After switching to `rpi_arm64_defconfig`, everything seems to work.

Using `rpi_arm64_defconfig`, we would gain support to the following boards, at the cost of having to rely on the firmware to provide the DTB: Raspberry Pi 3b - Raspberry Pi 3b+ - Raspberry Pi 4b - Raspberry Pi 400 - Raspberry Pi CM 3 - Raspberry Pi CM 3+ - Raspberry Pi CM 4 - Raspberry Pi zero 2 w

The only issue I can see with this approach is that we would have to rely on the firmware to apply multiple overlays to change the hardware configuration (enable USB, disable Bluetooth, ...). That would not be a blocker for me, but maybe it would be for you. 

For context, I got interested in the project because I would like my [boot2container](https://gitlab.freedesktop.org/mupuf/boot2container) initramfs to ship a single kernel/initrd that would be bootable on as many AARCH64 SBCs as possible without needing me to create custom kernels and rootfs per board. Additionally, i would really like to be able to netboot my SBCs over HTTPS using iPXE (like I do for my x86_64 computers). So, coming from an x86_64 world, I fully agree with you guys that it is time to for the ARM world to start separating the different functions of the firmware (hardware init, hardware config, and bootloader).

What are your thoughts?

Thanks to @craftyguy from PostmarketOS asking me to look into tow-boot more when it wouldn't boot on my CM4.

oneingan, 2026-09-02:
I can now confirm the version-skew concern described here on a Raspberry Pi 3 Model B.

### Environment

- Raspberry Pi 3 Model B (`aarch64`)
- Tow-Boot `release-2023.07-007` / U-Boot 2023.07
- systemd-boot 261.2
- NixOS with upstream/mainline kernels 6.12.107 and 7.2.2

The Tow-Boot firmware FAT used for this test has no external Pi 3 DTB, so without a `devicetree` line in the BLS entry the kernel receives the firmware-provided/mangled tree.

### A/B result

| Kernel | Device Tree | Result |
|---|---|---|
| 6.12.107 | firmware-provided DT | boots |
| 6.12.107 | matching upstream kernel DTB | boots |
| 7.2.2 | firmware-provided DT | hangs, reproduced twice |
| 7.2.2 | matching upstream kernel DTB | boots, reproduced twice |

With 7.2.2 and the firmware-provided tree, the kernel mounts the real root and reaches early userspace, but the whole machine stops making progress at approximately 42 seconds. The persistent journal ends while local filesystems and tmpfiles are being initialized, before the SMSC95xx USB Ethernet device is enumerated. There is no panic, OOM report, or explicit MMC error in the journal. An initially enabled 15-second hardware watchdog caused repeated resets; after disabling it, the same underlying condition remained as a permanent hang.

The fix was to have systemd-boot load the DTB built with the selected kernel, for example:

```nix
hardware.deviceTree = {
  enable = true;
  name = "broadcom/bcm2837-rpi-3-b.dtb";
  filter = "bcm2837-rpi-3-b.dtb";
};
```

This produces a BLS entry containing:

```text
devicetree /EFI/nixos/...-bcm2837-rpi-3-b.dtb.efi
```

Board-specific identity properties normally injected by the Raspberry Pi firmware were restored separately in a build-time DT overlay.

With the matching 7.2.2 DTB, the board completed two clean boots, enumerated SD and USB Ethernet normally, and passed a synchronized 64 MiB filesystem write/read test. The userspace configuration was otherwise unchanged.

This does not identify one individual bad property, but the controlled kernel/DT matrix strongly indicates an incompatibility between the firmware-provided Pi 3 tree and newer mainline kernels. It appears to be the Pi 3 equivalent of the risk anticipated in this issue.

Would it make sense to document that an OS boot loader should provide a kernel-matched upstream DTB on `raspberryPi-aarch64`, or to cover this through the DTB work in #192 / #327?