Tow-Boot/Tow-Boot #76: allwinner: Suppor eMMC boot partitions

samueldr, 2022-01-23:
> [smaeul] samueldr: allwinner does support boot partitions https://u-boot.readthedocs.io/en/latest/board/allwinner/sunxi.html#installing-on-emmc-on-board-flash-memory

> [smaeul] according to https://patchwork.ozlabs.org/project/uboot/cover/20210712100651.6912-1-andre.przywara@arm.com/ it is before the user data partition

 - https://u-boot.readthedocs.io/en/latest/board/allwinner/sunxi.html#installing-on-emmc-on-board-flash-memory
 - https://patchwork.ozlabs.org/project/uboot/cover/20210712100651.6912-1-andre.przywara@arm.com/
 - https://linux-sunxi.org/Bootable_eMMC

With this, supporting *dedicated storage* for Allwinner boards (range to be confiremd, but H3, and A64 at the very least) should be possible through the eMMC boot partitions.

Writing and managing should be pretty much equivalent to SPI installation.

Note that for some boards, eMMC is not *fixed* storage. When a board supports SPI, we shouldn't build the eMMC boot partition support in official builds, it provides no real benefit, especially when boards often treat eMMC as non-fixed storage.

RadxaYuntian, 2022-01-24:
Same is true for (at least our) Amlogic devices. Some SKU of Radxa Zero (Amlogic S905Y2) and Radxa Zero 2 (Amlogic A311D) contain onboard eMMC with boot partition support. They belong to meson-g12 family.

samueldr, 2022-01-24:
I know much less about Amlogic devices, my personal sample size of one (1) s805x device.

Though I spy useful notes about the startup sequence here:

 - https://wiki.radxa.com/Zero/install/Boot_Troubleshooting

RadxaYuntian, 2022-01-24:
Want to join our [discord](https://rock.sh/go) so we can have some talk about this project and more?

closed 2022-03-05 by samueldr