Tow-Boot/Tow-Boot #79: [feature request] boot order saving

13werwolf13, 2022-01-27:
first, I want to thank you for the work done. I’m experimenting a lot now to help opensuse maintainers fix many things in the PineBook Pro distribution and I had to disassemble my device several times a day until I was advised by Tow-Boot.

and now about the main thing :
I really miss the opportunity to change the order of the boot devices to be sure that without manual actions the load will come from where I want (this can be really useful for many). tell me if it is possible to add this functionality?

samueldr, 2022-01-27:
Hi :wave:

It is possible, and planned.

reukiodo, 2024-07-17:
I'd like to +1 this request.

In the meantime, for those of us that have towboot on SPI, is there a quick way to prioritize external boot media over internal ? I'm thinking something like setting the default order to: USB > uSD > NVMe > eMMC > netboot > bootmenu