Tow-Boot/Tow-Boot #81: feature request: u-boot should support extlinux.conf.once for a 'bootnext' experience

colemickens, 2022-01-27:
`sudo efibootmgr --bootnext XXXX` allows, via nvram, a one-time boot override to be specified.

On many Tow-Boot/u-boot platforms nvram isn't available, and something like extlinux is being used. I think grub has non-nvram mechanisms for doing this as well.

The request comes down to u-boot looking for an extra extlinux.conf file, maybe named `extlinux.conf.once` or `extlinux.conf.bootnext` or some such. U-boot would read that file, delete it, then (try to) boot it.

colemickens, 2022-01-27:
To elaborate, I am interested in automating rpi updates, and validating them.

I think I have most of a strategy down to test u-boot upgrades, but I also want to be able to test both the mainline and foundation kernels automatically. My Pi config default boots mainline, but has the ability to boot a "Foundation" specialisation. This issue's requested functionality would allow me to programmatically hack up a Foundation-specific extlinux conf and then boot it a single time for validation purposes. (Rather than the alternative, which is to interactive catch u-boot and pick the specialisation boot).

Another alternative to this would be having my testing automation simply change the kernel in use and re-deploy but that's not ideal and this functionality is generally useful.

samueldr, 2022-01-27:
Functionality could be generally useful to allow the operating system to mark a new "unknown if bootable" generation bootable on the first boot.

That OS could update the boot entries list, keep the current booted generation as active and default, add a `.once` entry. When booting the new generation, the OS would have an built-in way (hand-waves) to update the boot entries list with the new one as good.