Tow-Boot/Tow-Boot #82: feature request: (hack) u-boot passes its version as a kernel parameter

colemickens, 2022-01-28:
As discussed in "NixOS on ARM" room, and related to my goal of automated testing of new u-boot builds, I would like to know what build of u-boot was used to boot the kernel.

The feature request is a hack, inferior to passing information along via DTB, but is probably more easily implemented today, depending on the u-boot extlinux handling code.

This would allow me to install a new u-boot build, "tryboot" it on rpi4, and then know from the booted system that I succesfully booted via the new build, rather than being booted because the WD kicked in and then the normal u-boot booted.

colemickens, 2022-01-28:
I think this variable: https://source.denx.de/u-boot/u-boot/-/blob/761a1786e125ce0e2f472f25f2b841d5f4e2f0cb/env/Kconfig#L822

Could be appended in here: https://source.denx.de/u-boot/u-boot/-/blob/761a1786e125ce0e2f472f25f2b841d5f4e2f0cb/boot/pxe_utils.c#L516

samueldr, 2022-02-09:
As a kernel parameter, I don't like the idea.

Through other means, I'm all for it.

I wonder if the DT spec should even specify such a list of properties. I know *EFI* itself has some way to do it, `efi: EFI v2.40 by American Megatrends` is logged at `0.00000` during boot. I know U-Boot provides a string for that, though IIRC it's not the actual version used, but a "compatible" version number.

samueldr, 2023-10-18:
Since then, we do provide DMI information. It should be sufficient.

There's also uSWID support if scanning the blob is an acceptable alternative.

(Do feel free to elaborate if it's not sufficient.)

closed 2023-10-18 by samueldr