Tow-Boot/Tow-Boot #88: Pinebook Pro + Tow-Boot 2021.10-003 = infloop on boot

zackw, 2022-02-04:
I copied `pine64-pinebookPro-2021.10-003/spi.installer.img` to an SD card and booted my PBP from it.  Immediately after the old bootloader transfers control to the SD card, it goes into an infinite loop printing text to the screen.  It goes by too fast for me to read it properly, so I tried to record it with my phone, which came out really blurry, but I think it's saying something like this:

```
*********
No menu system available
*********
Use `[illegible]` to erase the SPI.
Use `[illegible]` to erase and flash to SPI.
Use `reset` to reset the system.
Unknown command `lb-menu` - try `help`
Unknown command `bootmenu` - try `help`
```

This message repeats until I hard-poweroff the laptop.

samueldr, 2022-02-04:
This is an issue that needs solving, "Using the U-Boot based installer from a non-Tow-Boot build of U-Boot".

Meanwhile, a work around is to [neuter your current U-Boot install (I suppose on eMMC)](https://github.com/Tow-Boot/Tow-Boot/issues/47). This would mean that Tow-Boot would be started, and it would have the `tb-menu` command to correctly show the menu.

A possible long-term solution will be to do like we're aiming to do for the "blind" phone UX: have a Linux system that is more easily controllable than the platform firmware install Tow-Boot. Another option would be to be extremely mindful of other U-Boot configs and try to figure out a generic solution. Though since SPI Flash (sf) support may not even be present, it's pretty much a moot point.

Strit, 2022-02-06:
This also happens on 001 and 002.

So it's because the current flashed u-boot is missing an option, that lets Tow-boot show it's menu?

samueldr, 2022-02-06:
Yes. Exactly.

I *thought* I had handled this correctly in the script, but I guess I hadn't. I have to re-test against stock U-Boot at some point.

zackw, 2022-02-07:
Neutering the current install (yes, on eMMC) worked.  This is a scary thing to ask people to do before installing Tow-Boot, though, since it may render the computer unbootable.  I wouldn't have been willing to risk it if I didn't have a second SD card with a recovery environment that I already knew worked.

I'm also surprised this is a problem -- it means the installer doesn't work the way I thought.  I expected the U-Boot on the eMMC would *chainload* the SD card, and therefore the installer would be independent of whatever was on the eMMC.  It sounds like instead what happens is the installer's scripts are being run by the existing U-Boot?

samueldr, 2022-02-07:
U-Boot itself does not mandate any U-Boot install can *chainload* any other U-Boot install. In my experience, none of the platforms seem to be able to do so.

And your understanding is pretty much spot-on: the installer scripts are used as-they-are by currently running U-Boot.

Though note that, thinking about it a bit more, even if it was handled correctly, it's possible that the `sf` command wouldn't be configured to work, and thus it would be unusable to flash anyway. This may all be side-steppd with a specialized Linux distro, which might *anyway* be required to do diagnostics (e.g. identify non-Tow-Boot platform firmwares, and other useful things).

curiositry, 2023-06-01:
@zackw For anyone else who has this issue (I did too), I was able to get the text stop scrolling with "Ctrl+C" (and then issue  commands).

port19x, 2023-06-10:
#47 did the trick for me :+1: