Tow-Boot/Tow-Boot #90: [suggestion] Add spi image to pinebook-a64 package

Strit, 2022-02-06:
According to the [pine64 wiki page on the Pinebook](https://wiki.pine64.org/wiki/Pinebook#Datasheets_for_Components_and_Peripherals) the Pinebook A64 has an 128 Mb SPI NOR flash chip. But the pinebookA64 package in the release section does not contain an spi-imager.img.

I would like to see it added. Thanks for this awesome project.

samueldr, 2022-02-06:
I must assume this came erroneously from copying the list from another of their A64 platforms. [The change in question](https://wiki.pine64.org/index.php?title=Pinebook&diff=2316&oldid=2312).

SPI Flash [is not present in the schematics](https://files.pine64.org/doc/pinebook/pinebook_mainboard_schematic_3.0.pdf). Neither has it been added [to the device tree source](https://github.com/torvalds/linux/blob/master/arch/arm64/boot/dts/allwinner/sun50i-a64-pinebook.dts). ([Here's the equivalent part for SOPINE/A64-LTS](https://github.com/torvalds/linux/blob/master/arch/arm64/boot/dts/allwinner/sun50i-a64-sopine.dtsi#L52-L62)).

It is "common knowledge" that the Pinebook A64 doesn't have SPI Flash... Which means there's still chances there is SPI Flash and no one bothered to look :/.

**But**, even if there's no SPI Flash available, there might be a way forward for an almost equivalent installation method. Stay tuned.

Strit, 2022-02-06:
That makes sense then. To bad, had hoped for an SPI solution on the Pinebook.

closed 2022-02-06 by Strit

samueldr, 2022-02-06:
Don't worry, I was pretty annoyed at that too, and was glad to see it present on the Pinebook Pro.