Tow-Boot/Tow-Boot #103: Flashing to Odroid-N2+ SPI fails

mth, 2022-02-18:
Tried to flash the 2021.10-003 release to SPI on 4GB Odroid-N2+.

* Wrote the installer to SD card with dd
* Booted the SD card
* Selected `Flash firmware to SPI` (from menu)

This failed with `SF: 8388608 bytes @ 0x0 Read: ERROR -110`.

Then tried `Completely erase SPI`, this failed with "No SPI flash selected. Please run `sf probe' ".

samueldr, 2022-02-18:
@strit recently faced similar, or the same, issue.

I don't have the hardware on hand to check. Though it has to be noted that SPI and eMMC shares something on the bus, which may be causing issues.

cc @tpwrules and @L-as

tpwrules, 2022-02-18:
It has detected the board fine, see `Board variant: n2-plus` on line 117 of the log. But in any case, I also have this problem. I booted with the SD card in, eMMC removed, and the boot switch set to MMC, then tried to flash from the menu and got this same error.

Commit 4425d07 which introduces N2+ support to Tow-Boot 002, flashes and boots from SPI correctly though following those same steps. I wonder if upstream 2021.10 broke something. I might be able to debug next week.

samueldr, 2022-02-18:
Thanks for confirming @tpwrules, I guess yeah this could be bisected. I should look into ensuring the build tooling works without patches to build a stock U-Boot.

tpwrules, 2022-02-18:
I did some bisecting and U-Boot commit e2e95e5e breaks `sf probe`. They realized this eventually and U-Boot commit b81ce79d fixes that but `sf read` still doesn't work properly. My next step would be to figure out how the SPI controller was configured pre-e2e95e5e and why it's not that way post-b81ce79d but that will have to wait until next week sometime.

tpwrules, 2022-02-19:
I lied again, oops. Please give this patch a try.

L-as, 2022-02-19:
FYI flashing from Linux worked flawlessly for me as a workaround.

knownunown, 2022-02-22:
FYI (if you don't have a SD card with Linux handy) it's also possible to flash Tow-Boot "manually" from the U-Boot console. All you have to do is to bring up the SPI flash with the correct clock speed:
```
sf probe 0 52000000 0
```

mth, 2022-02-23:
> FYI flashing from Linux worked flawlessly for me as a workaround.

Succeeded with flashcp to mtd0 (after choosing spi capable dtb in armbian). :)

samueldr, 2022-08-19:
Oh, this is now fixed with #149 I believe.

closed 2022-08-19 by samueldr

L-as, 2022-08-19:
Great!