Tow-Boot/Tow-Boot #104: doc: request to add device list and build instructions

tpwrules, 2022-02-18:
It would be useful to have a prominent list of the supported boards and their Tow-Boot names somewhere on the website.

I'd also like to see a reminder of the build instructions (i.e. `git clone ...`, `nix-build  -A ...`). Whenever I come back to this project it takes me too long to figure out what exactly I need to ask Nix for. Modifying the "warning: you're about to build all the boards" to have a link to the page above that lists the supported boards and names would be helpful too.

samueldr, 2022-02-18:
+1, anyway we need a list of "supported" devices, and a way to provide per-device information, not unlike what Mobile NixOS automatically does.

samueldr, 2022-02-18:
Relatively relevant:

```
~/tmp/u-boot/u-boot $ nix-shell ~/Projects/Tow-Boot/Tow-Boot -A pine64-pineA64LTS.config.Tow-Boot.outputs.firmware
```

This can be used to have a shell with the required environment.

samueldr, 2022-03-05:
*Incomplete* in #67; a device list was added.

In the future build instructions can be added. I want to think a bit more about it, as I don't want any focus on these. It's not *expected* that users will build. So I don't know how to expose that in the documentation.