Tow-Boot/Tow-Boot #110: ENH: "Chainload" to {grub,}

westurner, 2022-02-22:
Is it already possible to "chainload" (or better) into grub; where I already have:

- a (distro) managed menu of initrd/kernel combos 
- password auth
- an interface for editing the boot entry to add e.g. "rescue" for systems
- a perfectly good EFI shim as needed

- "(UEFI) Chainloading GRUB from GRUB"
  https://unix.stackexchange.com/questions/343632/uefi-chainloading-grub-from-grub
- https://www.gnu.org/software/grub/manual/grub/html_node/Chain_002dloading.html

samueldr, 2022-02-22:
"chainload"ing not necessarily. It depends exactly what you mean here.

U-Boot, and thus Tow-Boot, has a minimal UEFI implementation, but it's proper enough to execute the GRUB UEFI build.

Thus, you can execute the UEFI GRUB program without issue to select boot entries, and edit boot entries. This is even the actual goal for full EBBR compliance, rely mainly on UEFI.

"chainload"ing would also be supported in that case, since any EFI program can execute any other EFI programs. This was verified to work recently by verifying that [rEFInd](https://www.rodsbooks.com/refind/) works. With rEFInd installed to the fallback boot location, Tow-Boot will boot rEFInd just fine, and rEFInd will scan all available media just like it does on `x86_64` and provide options for booting into any detected EFI programs. From there it's detecting the bootloader from the distro I'm using (GRUB) and can start it as expected.

So, keeping this to a superficial interpretation, it's already possible. (With the caveat that input/display needs to be supported by Tow-Boo/U-Boot to be useful.)

Though I suppose there might be a more involved need than just executing an EFI program from another EFI program. Can you explain the exact scenario you're looking for?

westurner, 2022-02-22:
Use case: [Fedora] already manages my grub cfg menu entries.

- https://github.com/bengtfredh/pinebook-pro-fedora-installer/issues/27
- https://news.ycombinator.com/item?id=30416100 :
  > - [ ] ENH: Could there be bootloader game(s) with progress bars for
e.g. teh systemd boot sequence? Obvs, bootloader game high scores can't be
recorded until further into the boot sequence

samueldr, 2022-02-22:
Installing Fedora wtith the upstream UEFI iso already works (assuming the device is supported by their mainline images).

Bootloader games or anything else is squarely out of scope for this project. It would need to be handled by the bootloaders themselves. At that point you should think of it as running UEFI applications, under an abstract UEFI runtime.

Note that the Hacker News thread you linked to is about the *legacy x86_64* boot scheme. It is generally not relevant for targets supported by Tow-Boot/U-Boot.

westurner, 2022-02-22:
- when your one (1) bootloader config fails, with tow-boot, can I fix it
without another boot media?

- Grub allows editing the boot entry before running it
- Linux Distro maintains a 'rescue' grub menu entry
  https://man7.org/linux/man-pages/man1/systemd.1.html ctrl-f rd.rescue

- Use Case: Interactively develop and test additional bootloader entries,
such as {games, grub, tow-boot configs} instead of or during normal boot

westurner, 2022-02-22:
But again, many OSes already maintain grub.cfg for you; for example pon
kernel upgrade.

Supporting the existing config or just booting to grub would be ideal

https://github.com/rhboot/grubby/blob/master/grubby.c

samueldr, 2022-02-22:
You can already directly use GRUB. GRUB will work as usual, including editing boot entries and having a rescue entry.

westurner, 2022-02-22:
DOC: README: How to launch into Grub from tow-boot

... Might've been a better issue title then

samueldr, 2022-02-22:
I would say that the [upstream documentation](https://u-boot.readthedocs.io/en/latest/develop/uefi/uefi.html) document the UEFI implementation, but it's not useful really.

The gist of it is:

There is no EFI vars support at this time. This means that saving boot options using `efibootmgr` won't work. The solution / workaround is to install the UEFI program to execute at the default fallback location in the ESP.

So, to boot GRUB, make sure your distro installs it, or a shim, at `ESP:\BOOT\EFI\bootaa64.efi` assuming this is for AArch64.

westurner, 2022-02-22:
But it can't be a symlink because EFI boot partitions must use FAT, and
\_FAT doesn't support symlinks?

Could that be relatively static shim, as well , so that cp'ing after every
grub update is unnecessary?

samueldr, 2022-02-23:
I have not talked about symlinks. Symlinks are not needed.

By shim, I'm talking about a common concept where a small executable loads another executable. "The" [shim from red hat](https://github.com/rhboot/shim) is an example of one.

At this point there is nothing specific to Tow-Boot in this discussion. All of this is pure UEFI.

We have limited support capabilities in this project, questions about using UEFI should be directed to your distribution, or to more general support channels around booting.

westurner, 2022-02-23:
So you would advise to:

- un-flash tow-boot? `dd` with cached args?
- initially, copy the efi shim executable to the FAT EFI partition
- on every update to grub, copy the efi shim executable to the FAT EFI
partition

My mistake; IDK why I assumed there'd be an advantage to flashing tow-boot
(or u-boot, or pboot) with a config that launches the grub efi shim located
at the distro default location

Thanks for the magical path

samueldr, 2022-02-23:
There is no one answer for uninstalling Tow-Boot. Generally speaking it will be "erase the block storage where it's stored". It might be SPI, mmcboot, or (less likely) a partition on a storage media.

> My mistake; IDK why I assumed there'd be an advantage to flashing tow-boot (or u-boot, or pboot)

I was under the assumption that you are targeting an AArch64 system. But given what you're saying, I'm beginning to doubt. U-Boot, Tow-Boot and p-boot are early startup software generally used for ARM systems, and wouldn't be used for your "standard" x86_64 (intel/AMD) computer. They play a role closer to what the UEFI firmware or BIOS would on these computers.

Maybe I should have asked first what device this was about, but I did not as the answer would be the same.

westurner, 2022-02-23:
This is for a Pinebook Pro.

A bit more searching found this for uboot: 
"Installing and running grub on rock64 (rk3328) arm64" 
https://gist.github.com/tstellanova/dea7593a7dfe4f48432a58cb007e7056 :

***
Starting with Rock64, u-boot installed to SPI flash, Debian stretch installed to eMMC or micro SD card.

## Setup /boot
Add things like dtb and any images you want to eventually load with grub

## install grub
From Debian:
```
sudo apt-get -y install grub2-common efivar grub-efi-arm64
sudo grub-install /dev/mmcblk1p6 --target arm64-efi
```
Should now have `/boot/efi/debian/grubaa64.efi`

## grub boot
- interrupt autoboot, drop into uboot command line
```
load ${devtype} ${devnum}:${distro_bootpart} ${kernel_addr_r} efi/debian/grubaa64.efi
load ${devtype} ${devnum}:${distro_bootpart} ${fdt_addr_r} dtb
bootefi ${kernel_addr_r} ${fdt_addr_r}
```

***

- [ ] DOC: fedora: https://github.com/Tow-Boot/Tow-Boot/blob/development/doc/distributions/fedora.md
  - [ ] "ENH: Graphical boot menu: TowBoot, UBoot, SPI flash" https://github.com/bengtfredh/pinebook-pro-fedora-installer/issues/27
    - https://github.com/bengtfredh/pinebook-pro-fedora-installer/blob/master/fedora-installer#L206 generates extlinux.conf

samueldr, 2022-02-23:
I already stated what is needed for booting from UEFI.

Make sure your distro installs a shim or the bootloader to the default fallback location, and U-Boot or Tow-Boot will pick it up.

No need for dropping to the U-Boot command-line.

Frankly, I'm getting more and more confused about what you're saying here.

If you want to get support, please state **what you have tried** and how it failed, if it failed.

Please don't create elaborate hypothetical scenarios and what-ifs. Please write in complete english sentences. Please don't make spurious non-actionable "TODO" lists.

closed 2022-02-24 by samueldr