Tow-Boot/Tow-Boot #114: pine64-pinephoneA64: When the phone is powered off, plugging a USB-C cable doesn't power it on

tomfitzhenry, 2022-03-06:
Context: Found while testing the development branch, as discussed in Matrix.

### Steps to reproduce

1. Use commit fc6ca208bf03f17438794f7497b52bbc41f5d42b.
1. Install Tow-Boot to MMC Boot of a Pinephone.
1. Power phone off.
1. Plug USB-C cable in.

### Expected

Phone powers on and boots.

### Actual

Phone does not power on.

### Notes

This behaviour preserves the stock Pinephone behaviour, which is useful for development purposes to power on when the USB-C cable is plugged in (since it allows powering on the device when the case is removed).

But maybe users don't expect the phone to boot on cable entry?

samueldr, 2022-03-06:
AFAIK this happens outside of the purview of the platform firmware.

AFAICT [it is not possible to configure as desired](http://files.pine64.org/doc/datasheet/pine64/AXP803_Datasheet_V1.0.pdf).

Under 9.1.1 it is stated that charger insertion triggers a power-on. `00H` is not a configuration register, but it can be read to know if the device was powered-on via the charger.

But, *I can't reproduce*.

 1. disconnect charger
 2. remove battery
 3. insert battery
 4. connect charger

The phone starts.

Considering this is done from total power-off, AFAIK the PMIC is reset in its default state. I might have missed something from a cursory search in the datasheet, maybe it's configurable and being confgured by the driver in the OS you're running. (I'm most likely not running the same kernel you are. Yes, even with Mobile NixOS.)

tomfitzhenry, 2022-03-06:
> But, I can't reproduce.

I have just tried, and I believe this was due to me using the https://pine64.com/product/pinephone-pinephone-pro-pindio-lora-add-on-case/ case (which I had used for the first time), which was more difficult to plug in than the stock case.

Sorry for wasting your time!

closed 2022-03-06 by tomfitzhenry