Tow-Boot/Tow-Boot #116: Document MMC Boot in getting-started.md

tomfitzhenry, 2022-03-06:
Context: Found while testing the development branch, as discussed in Matrix.

MMC Boot (`mmcboot.installer.img`) should be documented in https://github.com/Tow-Boot/Tow-Boot/blob/released/doc/getting-started.md.

closed 2022-03-06 by samueldr

Shaolin-cpu, 2023-04-29:
I can't find any reference anywhere about where to find or how to generate `mmcboot.installer.img`.

quinn-dougherty, 2023-08-07:
I find this very confusing, cuz there's no `nix-build` command in the docs that will give me `mmcboot.installer.img` in a `result`, and I extracted / untar'd some downloads on the latest release tag and none of them contained `mmcboot.installer.img`.

samueldr, 2023-08-08:
The file will only be present for systems where MMC boot is possible.

As of now, it needs to be a system with eMMC (no, SD cards won't work), and either Amlogic or Allwinner.

For which board/device did you intend to use that?

PR #117 (2022-03-06): doc: Updates for mmcboot
Fixes #116.

cc @tomfitzhenry