Tow-Boot/Tow-Boot #121: 2021.10-004 pine64-pinephonePro: flash to SPI -> "unrecognized JEDEC id bytes: ff ff ff ff ff ff"

tomfitzhenry, 2022-03-06:
I have a PinePhone Pro Developer Edition. I am able to boot to the Tow-Boot 2021.10-004 installer via SD card, but when I click to flash to SPI, it fails with a red screen displaying:

```
Error: /dev/mtdblock0 not found
$ dmesg | grep -i spi
---
[ ... ] spi-nor spi0.0: unrecognized JEDEC id bytes: ff ff ff ff ff ff
```

Any ideas? I see in the matrix chat that this has been discussed, but I couldn't see a fix/workaround mentioned.

tomfitzhenry, 2022-03-07:
Uh oh, I'm not sure. Is there a way I can easily check? https://wiki.pine64.org/wiki/PinePhone_Pro_Developer_Edition doesn't mention this.

PureTryOut, 2022-03-07:
If you're unsure, you have a regular DE :wink:

tomfitzhenry, 2022-03-08:
From that Tow-Boot matrix room history, the "ff ff ff ff ff ff" issue, for the Developer Edition, can be caused by `the presence of another U-Boot / platform firmware "holding" the SPI bus high` and `in your case, since I assume you have a DE variant, it would be helped by first neutering the firmware from the eMMC`.

More context, but that's already built into 004 release. https://github.com/samueldr/Tow-Boot/blob/66bb20672fc30224d732538a16c16ef610308871/boards/pine64-pinephonePro/0001-pine64-pinephonepro-device-enablement.patch#L2618-L2648

tomfitzhenry, 2022-03-23:
I tried Mobian's PinePhone Pro image to see if the SPI was recognised. I used [mobian-pinephonepro-phosh-20220306.img.gz](https://images.mobian.org/pinephonepro/weekly/mobian-pinephonepro-phosh-20220306.img.gz) and found that its dmesg showed that the spi chip was recognised.

[mobian-spi.txt](https://github.com/Tow-Boot/Tow-Boot/files/8329822/mobian-spi.txt)

    $ grep -i spi mobian-spi.txt 
    [    0.000000] GICv3: 256 SPIs implemented
    [    0.000000] GICv3: 0 Extended SPIs implemented
    [   10.130186] spi-nor spi0.0: gd25lq128e (16384 Kbytes)

tomfitzhenry, 2022-03-23:
I observed Tow-Boot over serial and noticed that it's booting the Tow-Boot off the MMC:

```
U-Boot SPL 2021.10 (Oct 04 2021 - 15:09:26 +0000)
Trying to boot from MMC1
NOTICE:  BL31: v2.5(release):
NOTICE:  BL31: Built : 00:00:00, Jan  1 1980

Tow-Boot 2021.10 (Oct 04 2021 - 15:09:26 +0000)003-pre [variant: noenv]
```

i.e. it's booting from MMC, and a pre-release that I installed as part of testing https://github.com/NixOS/mobile-nixos/pull/445#issuecomment-1010025425 in January.

This explains why gd25lq128e is not recognized: this patch was added in https://github.com/Tow-Boot/Tow-Boot/blob/b94838dfb8971cdeb841d3922051aaa8e108b085/boards/pine64-pinephonePro/0001-mtd-spi-nor-ids-Add-GigaDevice-GD25LQ128E-entry.patch .

If so, diagnosis: PEBKAC. :) :(

I'll test this by wiping eMMC, and actually booting from the SD image.

tomfitzhenry, 2022-03-23:
> I'll test this by wiping eMMC, and actually booting from the SD image.

Done. I was able to flash to SPI. :) PEBKAC.

I think I would have noticed this issue sooner if I'd have seen the version number on Tow-Boot's UI. I'll raise an issue suggesting that.

closed 2022-03-23 by tomfitzhenry

PureTryOut, 2022-03-23:
The main problem there is that u-boot has no display support yet on that device, so there is nothing to show the version number on

samueldr, 2022-03-23:
I think @tomfitzhenry implied in the serial output, rather than display.

Copying the log he shared:

```
U-Boot SPL 2021.10 (Oct 04 2021 - 15:09:26 +0000)
Trying to boot from MMC1
NOTICE:  BL31: v2.5(release):
NOTICE:  BL31: Built : 00:00:00, Jan  1 1980

Tow-Boot 2021.10 (Oct 04 2021 - 15:09:26 +0000)003-pre [variant: noenv]
```

Only in the "U-Boot proper" output is the version printed, and it's printed after the date!