Tow-Boot/Tow-Boot #126: Pinephone Pro bricked after tow-boot and failed write to internal storage

drgnomage, 2022-03-06:
I successfully loaded tow-boot installed onto a microSD card and installed it to the SPI on the PPP.I then used the hold volume up function to provide the "jumpdrive" feature so I could write to the internal storage.

Using Balena Etcher I wrote Manjaro Phosh to the exposed device but it failed part way through, I believe because of a dodgy cable.

Now when the device boots it just shows a red LED and vibrates continually.  Occasionally I can get it to behave like normal, hold vol up gets me a blue LED and hold vol down yellow LED but they don't function properly.  So it doesn't expose the internal story or boot from the microSD.

Any help would be greatly appreciated.

zen0bit, 2022-03-07:
I haven't luck with etcher, try dd command instead, that works for me (not messed with internal storage yet)...

drgnomage, 2022-03-07:
After leaving the PPP unplugged for a couple of minutes and 10+ attempts, I managed to flash Manjaro + Phosh onto it and it now seems stable.  However the reason I was using tow-boot in the first place is still not working, waking from suspend.  Is there anything I need to do to the OS to get it to work?

PureTryOut, 2022-03-07:
> However the reason I was using tow-boot in the first place is still not working, waking from suspend

Although nice that it works (normally), that's not the main reason you should be using Tow-Boot :wink:  Instead you should use it for full EBBR compliance, the great and "boring" booting experience and the possibility to boot UEFI-using distros (which hopefully most distributions will be in the near future).

Anyway you might have issues because of an (almost) empty battery. Make sure it's charged before you start flashing things.

> Now when the device boots it just shows a red LED and vibrates continually.

That just means it tried to boot an OS but failed. Makes sense since you said flashing Manjaro failed halfway through. In these cases it's often best to just reflash the distro.

Calling your device bricked even though Tow-Boot was running fine is a bit harsh, it was just as much bricked as your laptop would be with a broken OS on it :wink: I suppose this issue can be closed, please make a new one for your suspend issue although I'd check with your distribution first to make sure it's not a distro-specific issue.

drgnomage, 2022-03-07:
> Although nice that it works (normally), that's not the main reason you should be using Tow-Boot wink Instead you should use it for full EBBR compliance, the great and "boring" booting experience and [...]

While all that stuff is great, with the PPP I'm much more concerned about the fundamental issue of battery life at the moment!

> Anyway you might have issues because of an (almost) empty battery. Make sure it's charged before you start flashing things.

I think this may have been the real issue, I didn't have the battery in at all and was just relying on the USB to power the device.  I didn't think that would be an issue.

> That just means it tried to boot an OS but failed. Makes sense since you said flashing Manjaro failed halfway through. In these cases it's often best to just reflash the distro.

It shouldn't have been booting an OS at that stage, it should have been giving me time to select boot from microSD or expose the internal storage but it just immediately upon turning on showed a red LED (normal) and continually vibrated.

> Calling your device bricked even though Tow-Boot was running fine is a bit harsh, it was just as much bricked as your laptop would be with a broken OS on it wink I suppose this issue can be closed, [...]

It's more akin to my laptop BIOS not giving me the option to boot from an alternate source and having a broken OS.

So I'm guessing the issue I was really raising is that the boot up doesn't work consistently if a battery is not present on the PPP?

PureTryOut, 2022-03-07:
> it just immediately upon turning on showed a red LED (normal) and continually vibrated.

Ah I misunderstood, that sounds like a bug then. It should indeed have given you some time to press a button. I'm guessing it's indeed either an empty battery or it not being plugged in correctly and thus relying on USB power only.