Tow-Boot/Tow-Boot #127: Pinephone A64 USB Mass Storage does not work if USB cable plugged in after booting

Be-ing, 2022-03-07:
With the Pinephone and Tow Boot installed to the eMMC boot partition, USB Mass Storage mode only works if the USB cable is connected to both the Pinephone and the host when the Pinephone is turned on. Connecting the cable after Tow Boot has started does not work. Here is dmesg output from the host turning on the PinePhone with the USB cable connected, then replugging the cable:

```
[ 490.112659] usb 1-3: new high-speed USB device number 38 using xhci_hcd
[  490.242140] usb 1-3: New USB device found, idVendor=1f3a, idProduct=1010, bcdDevice= 2.15
[  490.242153] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  490.242158] usb 1-3: Product: Pinephone (A64)
[  490.242162] usb 1-3: Manufacturer: Pine64
[  490.242165] usb 1-3: SerialNumber: 92c0a1ba8c8a2c07
[  490.244992] usb-storage 1-3:1.0: USB Mass Storage device detected
[  490.245724] scsi host1: usb-storage 1-3:1.0
[  491.277140] scsi 1:0:0:0: Direct-Access     Linux    UMS disk 0       ffff PQ: 0 ANSI: 2
[  491.277943] sd 1:0:0:0: Attached scsi generic sg2 type 0
[  491.278369] sd 1:0:0:0: [sdb] 61071360 512-byte logical blocks: (31.3 GB/29.1 GiB)
[  491.278520] sd 1:0:0:0: [sdb] Write Protect is off
[  491.278526] sd 1:0:0:0: [sdb] Mode Sense: 0f 00 00 00
[  491.278697] sd 1:0:0:0: [sdb] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[  491.284548]  sdb: sdb1 sdb2
[  491.286090] sd 1:0:0:0: [sdb] Attached SCSI removable disk
[  501.570484] usb 1-3: USB disconnect, device number 38
[  501.578633] sd 1:0:0:0: [sdb] Synchronizing SCSI cache
[  501.578657] sd 1:0:0:0: [sdb] Synchronize Cache(10) failed: Result: hostbyte=DID_NO_CONNECT driverbyte=DRIVER_OK
[  501.650512] pcieport 0000:00:1c.0: pciehp: Slot(0): Link Down
[  501.650516] pcieport 0000:00:1c.0: pciehp: Slot(0): Card not present
[  501.650554] pcieport 0000:01:00.0: can't change power state from D3cold to D0 (config space inaccessible)
[  501.650620] pcieport 0000:02:04.0: can't change power state from D3cold to D0 (config space inaccessible)
[  501.650986] pcieport 0000:02:02.0: can't change power state from D3cold to D0 (config space inaccessible)
[  501.859809] xhci_hcd 0000:38:00.0: can't change power state from D3cold to D0 (config space inaccessible)
[  501.859821] xhci_hcd 0000:38:00.0: can't change power state from D3cold to D0 (config space inaccessible)
[  501.859841] xhci_hcd 0000:38:00.0: Controller not ready at resume -19
[  501.859843] xhci_hcd 0000:38:00.0: PCI post-resume error -19!
[  501.859845] xhci_hcd 0000:38:00.0: HC died; cleaning up
[  501.859862] xhci_hcd 0000:38:00.0: remove, state 4
[  501.859865] usb usb4: USB disconnect, device number 1
[  501.860502] xhci_hcd 0000:38:00.0: USB bus 4 deregistered
[  501.860671] xhci_hcd 0000:38:00.0: remove, state 4
[  501.860676] usb usb3: USB disconnect, device number 1
[  501.861517] xhci_hcd 0000:38:00.0: Host halt failed, -19
[  501.861529] xhci_hcd 0000:38:00.0: Host not accessible, reset failed.
[  501.861707] xhci_hcd 0000:38:00.0: USB bus 3 deregistered
[  501.863486] pcieport 0000:01:00.0: can't change power state from D3cold to D0 (config space inaccessible)
[  501.863496] pcieport 0000:02:02.0: can't change power state from D3cold to D0 (config space inaccessible)
[  501.866290] pcieport 0000:02:01.0: can't change power state from D3cold to D0 (config space inaccessible)
[  501.873622] pcieport 0000:02:00.0: can't change power state from D3cold to D0 (config space inaccessible)
[  501.882635] pci_bus 0000:03: busn_res: [bus 03] is released
[  501.891787] pci_bus 0000:04: busn_res: [bus 04-37] is released
[  501.894085] pci_bus 0000:38: busn_res: [bus 38] is released
[  501.894946] pci_bus 0000:39: busn_res: [bus 39-6a] is released
[  501.897903] pci_bus 0000:02: busn_res: [bus 02-6a] is released
[  504.645839] usb 1-3: new high-speed USB device number 39 using xhci_hcd
[  504.760251] usb 1-3: device descriptor read/64, error -71
[  505.400254] usb 1-3: device descriptor read/64, error -71
[  505.622205] usb 1-3: new high-speed USB device number 40 using xhci_hcd
[  505.737202] usb 1-3: device descriptor read/64, error -71
[  506.369111] usb 1-3: device descriptor read/64, error -71
[  506.475702] usb usb1-port3: attempt power cycle
[  506.855834] usb 1-3: new high-speed USB device number 41 using xhci_hcd
[  506.856011] usb 1-3: Device not responding to setup address.
[  507.059830] usb 1-3: Device not responding to setup address.
[  507.268180] usb 1-3: device not accepting address 41, error -71
[  507.382181] usb 1-3: new high-speed USB device number 42 using xhci_hcd
[  507.382351] usb 1-3: Device not responding to setup address.
[  507.587848] usb 1-3: Device not responding to setup address.
[  507.795671] usb 1-3: device not accepting address 42, error -71
[  507.795821] usb usb1-port3: unable to enumerate USB device
```

braewoods, 2022-07-19:
I also wanted to note that I noticed the same issue but I was only able to get it to work with my Ryzen 7 Pro 5750G by connecting it to a USB 2.0 only port. For some reason it really hates USB 3.0 even though these work normally with my Pinephone after booting into Linux.

rosbeef, 2022-09-01:
Same problem with pinephone pro
Maybe a problem of USB-C DP as there is a negotiation in the current delivery (way and power). Is disable "negociation" and fix voltage and current to a specific value could be done in twoboot Mass storage mode ?

I think the phone should never be like a power bank (less with the PPP battery).
I have a pinebookpro docking deck and it consume all my battery if the dock is not powered. Or at least only for specific devices.

rosbeef, 2022-09-08:
more datas : 
in usb mass storage (volume up pressed at boot ) i can connect to the computer, then open files like usb mass 
if i disconnect the cable it stay in usb mass storage ()
if i reconnect to the same usb port of my computer pressing the button (volume up) to go to usb mass storage, it directly reboot on emmc without taking volume up pressed in account.

```
U-Boot SPL 2021.10 (Oct 04 2021 - 15:09:26 +0000)
Trying to boot from SPI
NOTICE: BL31: v2.6(release):
NOTICE: BL31: Built : 00:00:00, Jan 1 1980

Tow-Boot 2021.10 (Oct 04 2021 - 15:09:26 +0000)004 [variant: spi]

SoC: Rockchip rk3399
Reset cause: POR
Model: Pine64 PinePhonePro
DRAM: 3.9 GiB
PMIC: RK8180 (on=0x80, off=0x08)
MMC: mmc@fe310000: 3, mmc@fe320000: 1, mmc@fe330000: 0
Loading Environment from SPIFlash... SF: Detected gd25lq128e with page size 256B
*** Warning - bad CRC, using default environment

In: serial
Out: serial
Err: serial
Model: Pine64 PinePhonePro
Net: No ethernet found.
starting USB...
Bus usb@fe380000: USB EHCI 1.00
Bus usb@fe3a0000: USB OHCI 1.0
Bus usb@fe3c0000: USB EHCI 1.00
Bus usb@fe3e0000: USB OHCI 1.0
scanning bus usb@fe380000 for devices... 1 USB Device(s) found
scanning bus usb@fe3a0000 for devices... 1 USB Device(s) found
scanning bus usb@fe3c0000 for devices... 1 USB Device(s) found
scanning bus usb@fe3e0000 for devices... 1 USB Device(s) found
scanning usb for storage devices... 0 Storage Device(s) found
Please press [ESCAPE] or [CTRL+C] to enter the boot menu.
on
UMS: LUN 0, dev mmc 0, hwpart 0, sector 0x0, count 0xe678000
```
here i'm in usb mass storage
then disconnect and reconnect usb cable

```
-off
switch to partitions #0, OK
mmc0(part 0) is current device
Scanning mmc 0:1...
Found /extlinux/extlinux.conf
Retrieving file: /extlinux/extlinux.conf
839 bytes read in 15 ms (53.7 KiB/s)
U-Boot menu
1: Mobian GNU/Linux 5.19-rockchip
2: Mobian GNU/Linux 5.17-rockchip
Enter choice: 1: Mobian GNU/Linux 5.19-rockchip
Retrieving file: /miniramfs-5.19-rockchip
5587921 bytes read in 254 ms (21 MiB/s)
Retrieving file: /vmlinuz-5.19-rockchip
9362244 bytes read in 416 ms (21.5 MiB/s)
append: root=/dev/mapper/calamares_crypt console=ttyS2,1500000n8 consoleblank=00
Retrieving file: /dtb-5.19-rockchip/rockchip/rk3399-pinephone-pro.dtb
70260 bytes read in 39 ms (1.7 MiB/s)
Uncompressing Kernel Image
Moving Image from 0x2080000 to 0x2200000, end=3a50000
Flattened Device Tree blob at 01f00000

Booting using the fdt blob at 0x1f00000
Loading Ramdisk to f499d000, end f4ef13d1 ... OK
Loading Device Tree to 00000000f4988000, end 00000000f499c273 ... OK

Starting kernel ...
```

Yiannis128, 2024-02-07:
Is there a fix for this? I can't figure out how to get passed this problem?

ArenM, 2024-02-08:
Poking around a little bit, I was able to get it work (with upstream u-boot on the master branch) by enabling the following config options:

```
CONFIG_SYS_I2C_SUN8I_RSB=y
CONFIG_DM_PMIC=y
CONFIG_PMIC_AXP=y
CONFIG_REGULATOR_AXP_USB_POWER=y
```

And adding the following patch:

```
diff --git a/arch/arm/dts/sun50i-a64-pinephone.dtsi b/arch/arm/dts/sun50i-a64-pinephone.dtsi
index b25e7913f55..4b519918b65 100644
--- a/arch/arm/dts/sun50i-a64-pinephone.dtsi
+++ b/arch/arm/dts/sun50i-a64-pinephone.dtsi
@@ -512,5 +512,6 @@
 };
 
 &usbphy {
+       usb0_vbus_power-supply = <&usb_power_supply>;
        status = "okay";
 };
```

This patch is somewhat hacky because megi's kernel tree (specifically the axp7688 driver, which isn't upstream) expects to manage the power supply enable/disable state instead of the sunxi usb driver.