Tow-Boot/Tow-Boot #129: Version reporting

DylanVanAssche, 2022-03-07:
Would be great if the version is easily retrievable from the firmware.

Current options:
- DMI property (UEFI-only), which would allow us to use standardized interfaces, so my preference.
- Adding a string `VERSION=2022.01-004`
- Modifying the reported version by dropping the date: `Tow-Boot 2022.01-004`

Would help us for https://github.com/fwupd/fwupd/issues/4294

DylanVanAssche, 2022-03-14:
After some talks with @hughsie from fwupd on IRC, a better approach would be appending a blob with SWID metadata.
See https://blogs.gnome.org/hughsie/2022/03/10/firmware-software-bill-of-materials/

This would result into:

- A vendor.ini file
- Python tool to convert this INI file into a blob
- Append/Add the blob to the firmware

samueldr, 2022-03-14:
[noting after a short discussion in the Matrix room]

At a glance uSWID seems a good way forward, if I understand it right.

MartijnBraam, 2022-03-17:
I think the DMI data is also required in any case, for https://github.com/fwupd/fwupd/issues/4403

samueldr, 2022-09-19:
Note to self: also include the wip SMBIOS and version reporting that reports more detailed Tow-Boot version.

DylanVanAssche, 2022-09-19:
# uSWID data generation

uSWID is a specification to describe software information like about the software, creator, distributor, compiler, etc.
See https://csrc.nist.gov/Projects/Software-Identification-SWID/guidelines

Example `uswid.ini` file for an eMMC boot partition with Tow-Boot 2021.10-005:

```
[uSWID]
tag-id = acbd84ff-9898-4922-8ade-dd4bbe2e40ba
software-name = Tow-Boot.mmcboot.bin
software-version = 2021.10-005
version-scheme = multipartnumeric+suffix
product = Tow-Boot
summary = An opinionated distribution of U-Boot
edition = 2021.10
revision = 005
persistent-id = org.tow-boot.tow-boot

[uSWID-Entity:TagCreator]
name = Tow-Boot
regid = tow-boot.org
extra-roles = Licensor,Maintainer,SoftwareCreator

[uSWID-Entity:Distributor]
name = postmarketOS
regid = postmarketos.org
```

- `tag-id` is the value of commit ID or something else, doesn't matter which ID. It is just a more detailed version info (not used on fwupd though)
- Feel free to change TagCreator and Distributor values, these are just examples
- Of course, the version ID, edition and revision needs to be updated for each release.

To 'compile' this file into a binary file which can be integrated in an image, you can use `python-uswid`: https://github.com/hughsie/python-uswid This tool can be used like this:

```
uswid --load uswid.ini --save uswid.bin
```

Once you have this file, you can append it to your Tow-Boot image:

```
dd if=uswid.bin >> Tow-Boot.img
```

With `hexdump -C` you can verify if everything is fine:

```
000c5fe0  69 6e 65 70 68 6f 6e 65  2d 31 2e 31 00 00 00 00  |inephone-1.1....|
000c5ff0  00 00 00 03 00 00 00 04  00 00 00 0e 61 74 66 00  |............atf.|
000c6000  00 00 00 03 00 00 00 0a  00 00 00 04 73 63 70 00  |............scp.|
000c6010  75 62 6f 6f 74 00 00 00  00 00 00 03 00 00 00 06  |uboot...........|
000c6020  00 00 00 00 66 64 74 2d  31 00 00 00 00 00 00 02  |....fdt-1.......|
000c6030  00 00 00 01 63 6f 6e 66  69 67 2d 32 00 00 00 00  |....config-2....|
000c6040  00 00 00 03 00 00 00 19  00 00 00 57 73 75 6e 35  |...........Wsun5|
000c6050  30 69 2d 61 36 34 2d 70  69 6e 65 70 68 6f 6e 65  |0i-a64-pinephone|
000c6060  2d 31 2e 32 00 00 00 00  00 00 00 03 00 00 00 04  |-1.2............|
000c6070  00 00 00 0e 61 74 66 00  00 00 00 03 00 00 00 0a  |....atf.........|
000c6080  00 00 00 04 73 63 70 00  75 62 6f 6f 74 00 00 00  |....scp.uboot...|
000c6090  00 00 00 03 00 00 00 06  00 00 00 00 66 64 74 2d  |............fdt-|
000c60a0  32 00 00 00 00 00 00 02  00 00 00 02 00 00 00 02  |2...............|
000c60b0  00 00 00 09 66 64 74 00  6c 6f 61 64 61 62 6c 65  |....fdt.loadable|
000c60c0  73 00 66 69 72 6d 77 61  72 65 00 64 65 66 61 75  |s.firmware.defau|
000c60d0  6c 74 00 64 61 74 61 00  65 6e 74 72 79 00 6c 6f  |lt.data.entry.lo|
000c60e0  61 64 00 63 6f 6d 70 72  65 73 73 69 6f 6e 00 61  |ad.compression.a|
000c60f0  72 63 68 00 6f 73 00 74  79 70 65 00 23 61 64 64  |rch.os.type.#add|
000c6100  72 65 73 73 2d 63 65 6c  6c 73 00 64 65 73 63 72  |ress-cells.descr|
000c6110  69 70 74 69 6f 6e 00 74  69 6d 65 73 74 61 6d 70  |iption.timestamp|
000c6120  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*

---> Here uSWID data starts, right after the end of the Tow-Boot image <---

000c6500  00 00 00 00 00 00 00 53  42 4f 4d d6 ba 2e ac a3  |.......SBOM.....|
000c6510  e6 7a 52 aa ee 3b af 01  17 00 f4 00 00 00 a9 0f  |.zR..;..........|
000c6520  65 65 6e 2d 55 53 00 50  ac bd 84 ff 98 98 49 22  |een-US.P......I"|
000c6530  8a de dd 4b be 2e 40 ba  08 f5 01 74 54 6f 77 2d  |...K..@....tTow-|
000c6540  42 6f 6f 74 2e 6d 6d 63  62 6f 6f 74 2e 62 69 6e  |Boot.mmcboot.bin|
000c6550  0d 6b 32 30 32 31 2e 31  30 2d 30 30 35 0e 02 05  |.k2021.10-005...|
000c6560  a6 18 32 65 75 53 57 49  44 18 37 78 25 41 6e 20  |..2euSWID.7x%An |
000c6570  6f 70 69 6e 69 6f 6e 61  74 65 64 20 64 69 73 74  |opinionated dist|
000c6580  72 69 62 75 74 69 6f 6e  20 6f 66 20 55 2d 42 6f  |ribution of U-Bo|
000c6590  6f 74 18 36 63 30 30 35  18 34 68 54 6f 77 2d 42  |ot.6c005.4hTow-B|
000c65a0  6f 6f 74 18 2f 67 32 30  32 31 2e 31 30 18 33 75  |oot./g2021.10.3u|
000c65b0  6f 72 67 2e 74 6f 77 2d  62 6f 6f 74 2e 74 6f 77  |org.tow-boot.tow|
000c65c0  2d 62 6f 6f 74 02 82 a3  18 1f 68 54 6f 77 2d 42  |-boot.....hTow-B|
000c65d0  6f 6f 74 18 20 6c 74 6f  77 2d 62 6f 6f 74 2e 6f  |oot. ltow-boot.o|
000c65e0  72 67 18 21 84 01 05 06  02 a3 18 1f 6c 70 6f 73  |rg.!........lpos|
000c65f0  74 6d 61 72 6b 65 74 4f  53 18 20 70 70 6f 73 74  |tmarketOS. ppost|
000c6600  6d 61 72 6b 65 74 6f 73  2e 6f 72 67 18 21 81 04  |marketos.org.!..|
000c6610  04 80                                             |..|
000c6612
```

hughsie, 2022-09-20:
You might also want to use `--compress` when running `uswid` if you're worried about space; it should be a nice saving for repeated data like that.

PR #270 (2023-10-17): Tag output with uSWIDs
Fixes #129

* * *

With these changes, we now append an uSWID to the actual Tow-Boot "binaries", which in turn should unblock future work around fwupd.

```
~/.../Tow-Boot/uswid $ ./result-uswid/bin/uswid --load result/binaries/Tow-Boot.mmcboot.bin --save uswid.xml
Found USWID header at offset: 817665
```

```xml
<?xml version='1.0' encoding='utf-8'?>
<SoftwareIdentity xmlns="http://standards.iso.org/iso/19770/-2/2015/schema.xsd" xmlns:SHA256="http://www.w3.org/2001/04/xmlenc#sha256" xmlns:SHA512="http://www.w3.org/2001/04/xmlenc#sha512" xmlns:n8060="http://csrc.nist.gov [... cut]
closed 2023-10-17 by samueldr