Tow-Boot/Tow-Boot #131: PinebookPro unable to boot tow-boot installer.

Nik--, 2022-03-08:
I have U-Boot flashed to my SPI and Tow-Boot image on my SD Card. When I try to boot from within u-boot, it seems to find the boot file within the SD card, but it fails to load with the given problem:
[image]

benthetechguy, 2022-03-08:
The RK3399's hard-coded boot order is:
1. SPI
2. eMMC
3. SD Card

Normal U-Boot can't boot to the Tow-Boot SPI installer and thus get that exact error.
Your only solution is to erase the SPI. What risks do you speak of? There are basically none.

Nik--, 2022-03-09:
Well, reading the guide for flash erasing the Pinebook Pro, it looked like I've had to open the device and short circuit two pins to erase it. Therefore I thought that it was a risky thing to do. But apparently, there was another way, by installing `mtd-utils` package and running `flash_erase /dev/mtd0 0 0`, the SPI was erased. But for me that was still not enough, as the eMMC apparently had another u-boot in it, so I neutered it as well by running `dd if=/dev/zero of=/dev/mmcblk2 bs=512 seek=64 count=16`. In the end, the tow-boot installer managed to run and flash successfully to SPI.

closed 2022-03-09 by Nik--

benthetechguy, 2022-03-09:
Yes, that hardware method of shorting pins is meant to be a last resort for if you can't boot to an OS to run `flash_erase`.

myself600, 2022-11-10:
Yes,

`dd if=Tow-boot.spi.bin of=/dev/mtd0`

as mentioned [here](https://forum.pine64.org/showthread.php?tid=17386&pid=113301#pid113301) should do the work