Tow-Boot/Tow-Boot #134: Pine Phone Pro not able to install

kbmetz, 2022-03-10:
I've followed the directions to the letter 4 times, also I've tried 2 different SD cards.

Holding down the RE button and plugging in USB power, the phone vibrates and the LED comes on
Then the phone vibrates twice, and LED changes to yellow

Nothing but a blank screen after

Let the phone sit for 10 mins and never see a installer. Is this some strange bug?

Using pine64-pinephonePro-2021.10-004-pre

kbmetz, 2022-03-11:
Resolved my own issue. Needed to charge the phone overnight using the RE method.
It appears that tow-boot won't install if you are using the USB cable?

joshfleming, 2022-03-14:
I was able to install from SDcard while the USB cable was connected for charging. (My battery wasn't previously drained)

kbmetz, 2022-04-13:
Might be possible you are holding down the button too long?

mikkjant, 2022-04-13:
@magdesign 
I had the exact same problem. Just now I tried solution "after adding a volume label to the SD Card" found here:
https://forum.pine64.org/showthread.php?tid=15998&pid=108353#pid108353

So I formatted my SD card again with short volume label "test" - installed tow-boot again and after booting with this SD card tow-boot install was successful.

magdesign, 2022-04-14:
Thanks!

after the ~10th attempt I could flash Tow-Boot.
After `dd` the `spi.installer.img` to a 32GB SD card, I used GParted to rename the partitions to `test` and `test1`,  also resized the 2nd partition so there is no empty space left on the card. 

And finally it worked

PureTryOut, 2022-04-14:
This issue can be closed

Shermjj, 2022-05-08:
I had the exact same issue too. Didn't have to mess with the partition, just don't hold the RE button for too long when booting to SD.

senorsmile, 2022-05-14:
> I had the exact same issue too. Didn't have to mess with the partition, just don't hold the RE button for too long when booting to SD.

This was my issue as well.

silverspectro, 2023-08-10:
> I had the exact same issue too. Didn't have to mess with the partition, just don't hold the RE button for too long when booting to SD.

Exactly the same here. After reading this comment and trying messing with the partitions and all. I finally just `dd` the SDCard and tried to old the button ONLY untill the LED became yellow.
After 3 attempts (without reflashing, just trying to old the button differently) where the screen stayed black, it finally booted in the installation and worked as expected.

Thanks :)