Tow-Boot/Tow-Boot #137: PineNote port

phodina, 2022-03-22:
Hi,

I would like to help port Tow-Boot to PineNote. I do have Android and ArchLinux installed on the tablet.

However, I've never ported u-boot for new platform and would like to ask for guidance. On the other hand I'm familiar with Nix.

Are there any docs that I could read?

samueldr, 2022-03-22:
Hi! :wave: 

There might not be that much work to do, since I started working on that the past week-end. 

 - https://github.com/samueldr/Tow-Boot/tree/wip/pinenote

I'm basing the work off of this mainline-based work:

 - https://gitlab.com/pgwipeout/u-boot-quartz64

It will require setting a temporary config to use a local checkout of the fork:

```diff
--- a/boards/pine64-pinenote/default.nix
+++ b/boards/pine64-pinenote/default.nix
@@ -58,6 +58,10 @@ in
       })
     ];
 
+    uBootVersion = "2022.01";
+    useDefaultPatches = false; # XXX
+    src = builtins.fetchGit /Users/samuel/tmp/u-boot/pinenote;
+
     withLogo = false; # XXX
 
     builder.preBuild = ''
```

So the current state is *quite YMMV*. It also includes none of the Tow-Boot specific patches, and mainly re-uses the build infra. This is because I intend to "port" the patches to 2022.04 next.

Note that booting the supplied Android image is not supported, and is not intended to be supported. There is no reason to use the supplied Android image, and we don't even have the required sources (as a community) to build it and maintain it.

Anyway it's not like all that really matters. This is a "blind" boot UX, and one that can't use the phone UX due to lack of buttons and lack of feedback. The current implementation should not really change once the Tow-Boot patches are in.

For now, the expected usage will be:

 - Normal boot: do the distro boot command (as usual)
 - Boot with the magnet on the back "recovery" sensor, for now reboots in maskrom USB download mode, with all the caveats this implies. When USB gadgets will be working in U-Boot, this will likely be replaced with USB Mass Storage.
 - *Additional boot mode:* when a magnet is placed on the cover sensor, the device will pause and wait 2 seconds, if the magnet is removed after these two seconds, *the system will boot a `recovery.scr` script in the GPT partlabel `recovery` partition. I'm using this with a "Target Disk Mode" system which provides USB mass storage, but in practice this could be anything.

Note that I still need to do a writeup for the initial install. It's not exactly trivial. I ended-up copying the "shared" disk image, used `fallocate` to expand it some, added a `recovery` partition, and used `rkdeveloptool write 0 temp.img` to write the image. I don't really know if we'll have any easier method for initial install. At least for the time being.

samueldr, 2022-03-22:
> However, I've never ported u-boot for new platform and would like to ask for guidance. On the other hand I'm familiar with Nix.

Let's answer this *anyway* to help anyone who wants this question answered.

As you've observed, porting U-Boot is a prerequisite for Tow-Boot support, since Tow-Boot is a U-Boot distribution. Though I'm sorry, I don't have any specific knowledge here, as I'm building off of the hard work of more knowledgeable contributors. Targeting *actually new platforms* is likely a big job. Adding *new devices to well-supported platforms*, is likely easier though. 

I'm sorry I don't really have any guidance to offer around this particular subject.

(Not that it matters for the Pinenote, as knowledgeable individuals are doing the hard work of adding the RK356X platform, and adding the Pinenote to U-Boot.)