Tow-Boot/Tow-Boot #139: PineBook Pro - No display

ryanplusplus, 2022-03-27:
I've installed 2021.10-004 and 2021.10-003 to SPI and booted them from SD and USB (with SPI erased, eMMC disabled via the switch, and no other bootable devices present). In all cases I get a red power LED that indicates that Tow Boot is running, but the built-in display is just black (it's "lit", but doesn't show anything). I think I've even been able to navigate the menu blindly because I can make it reboot with the keyboard so I think the menu is "there", but the display simply isn't working.

Any ideas what I'm doing wrong?

merlink01, 2022-03-28:
As I know:
Tow Boot does not use a display.
If you need to see the output, you have to connect a serial connection.

On the PinephonePro I don't need a display, because there are only 3 options:
Do nothing = boot emmc
Vol Up = boot sdcard
Vol Down = expose emmc as an USB device

ryanplusplus, 2022-03-28:
@merlink01 I think the situation is different on the PineBook Pro. This video shows a menu on the PineBook Pro:
https://www.youtube.com/watch?v=wCVcjwpDCYo

samueldr, 2022-03-28:
@ryanplusplus that would be the first time I've seen reports of the sort, outside of a user replacing the panel with a "compatible", but different panel from another vendor.

I'm not sure what to do to help diagnose the situation, the display driver is an upstream U-Boot change, which while I initially was involved with early testing, I have not been involved in authoring.

ryanplusplus, 2022-03-28:
@samueldr thanks for the info. I bought the PBP secondhand so I guess it's theoretically possible that the display has an interesting history. Perhaps Pine64 has also changed vendors at some point.

ids1024, 2022-04-15:
https://en.opensuse.org/HCL:Pinebook-Pro-RK3399 suggests some sort of issue with 2021.10 on the PBP:

> Screen is not working with upstream u-boot version 2021.10. This is fixed in u-boot 2022.01

ryanplusplus, 2022-04-16:
> https://en.opensuse.org/HCL:Pinebook-Pro-RK3399 suggests some sort of issue with 2021.10 on the PBP:
> 
> > Screen is not working with upstream u-boot version 2021.10. This is fixed in u-boot 2022.01

Thanks! I guess I'll just sit tight until tow-boot syncs with 2022.01.

vincele, 2022-11-20:
Hello,
I cannot reproduce this.
TB 2021.10-005 on sdcard.
emmc disconnected.
spi empty.

I get the menu on display, and on serial console.

Do you also have this bug with the new version ?

ryanplusplus, 2022-11-20:
> Hello, I cannot reproduce this. TB 2021.10-005 on sdcard. emmc disconnected. spi empty.
> 
> I get the menu on display, and on serial console.
> 
> Do you also have this bug with the new version ?

I'm still able to reproduce this with the latest version. I only get a menu on the serial console with 2021.10-005 installed to SPI.

mhuesch, 2023-06-10:
I believe I am encountering the same issue? though I have tow-boot `dd`-ed to SD card. inserted that into PBP & booting from SD leaves the display black (not sure if is "lit" or not). I know that the display panel is functional b/c the original Manjaro OS shipped on the PBP could boot and show things on the display.

I'm using `Tow-Boot 2021.10 (Oct 04 2021 - 15:09:26 +0000)005 [variant: boot-installer]`

this is the GUI initialization error I see on the serial console:

```
/root # tow-boot-installer-gui
[display/drm_drv]: Starting DRM subsystem... (/dev/dri/card0)
[display/drm_drv]: error: drm device '/dev/dri/card0' does not support dumb buffers
[display/drm_drv]: error: modeset failed with error 95: Operation not supported

[display/drm_drv]: error: (falling back to fbdev...)
Error: cannot open framebuffer device: No such file or directory
[display/drm_drv]: DRM subsystem and buffer mapped successfully
WARNING: Error loading fallback font in FT_New_Face: 1
Error in FT_New_Face: 1
Error: Could not load font; falling back to built-in Roboto@22...       (/build/source/lvgui/hal.c #118)
libinput error: /nix/store/eeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee-libinput-aarch64-unknown-linux-gnu-1.16.4/share/libinput: failed to find data files
libinput error: Failed to load the device quirks from /nix/store/eeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee-libinput-aarch64-unknown-linux-gnu-1.16.4/share/libinput and /etc/libinput/local-overrides.quirks. This will negatively affect device behavior. See https://wayland.freedesktop.org/libinput/doc/1.16.4/device-quirks.html for details.
unable to add device to libinput context:: No such file or directory
Unable to add device to libinput context:: No such file or directory
libinput error: /nix/store/eeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee-libinput-aarch64-unknown-linux-gnu-1.16.4/share/libinput: failed to find data files
libinput error: Failed to load the device quirks from /nix/store/eeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee-libinput-aarch64-unknown-linux-gnu-1.16.4/share/libinput and /etc/libinput/local-overrides.quirks. This will negatively affect device behavior. See https://wayland.freedesktop.org/libinput/doc/1.16.4/device-quirks.html for details.
xkbcommon: ERROR: failed to add default include path /nix/store/eeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee-xkeyboard-config-aarch64-unknown-linux-gnu-2.33/etc/X11/xkb
[ERROR] (xkbcommon_init): error: Could not create xkb_context.
libinput error: /nix/store/eeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee-libinput-aarch64-unknown-linux-gnu-1.16.4/share/libinput: failed to find data files
libinput error: Failed to load the device quirks from /nix/store/eeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee-libinput-aarch64-unknown-linux-gnu-1.16.4/share/libinput and /etc/libinput/local-overrides.quirks. This will negatively affect device behavior. See https://wayland.freedesktop.org/libinput/doc/1.16.4/device-quirks.html for details.
xkbcommon: ERROR: failed to add default include path /nix/store/eeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee-xkeyboard-config-aarch64-unknown-linux-gnu-2.33/etc/X11/xkb
[ERROR] (xkbcommon_init): error: Could not create xkb_context.
libinput error: /nix/store/eeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee-libinput-aarch64-unknown-linux-gnu-1.16.4/share/libinput: failed to find data files
libinput error: Failed to load the device quirks from /nix/store/eeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee-libinput-aarch64-unknown-linux-gnu-1.16.4/share/libinput and /etc/libinput/local-overrides.quirks. This will negatively affect device behavior. See https://wayland.freedesktop.org/libinput/doc/1.16.4/device-quirks.html for details.
xkbcommon: ERROR: failed to add default include path /nix/store/eeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee-xkeyboard-config-aarch64-unknown-linux-gnu-2.33/etc/X11/xkb
[ERROR] (xkbcommon_init): error: Could not create xkb_context.
```

<details>

<summary>full transcript showing a reboot (click to unfold)</summary>

```
/root # reboot
The system is going down NOW!
Sent SIGTERM to all processes
Sent SIGKILL to all processes
[   26.022126] dw-mipi-dsi-rockchip ff960000.mipi: failed to write command FIFO
[   26.030062] panel-himax-hx8394 ff960000.mipi.0: Failed to enter sleep mode: -110
[   26.038367] panel-himax-hx8394 ff960000.mipi.0: Failed to disable panel: -110
[   26.048683] reboot: Restarting system

U-Boot TPL 2021.10 (Oct 04 2021 - 15:09:26)
Channel 0: LPDDR4, 50MHz
BW=32 Col=10 Bk=8 CS0 Row=16/15 CS=1 Die BW=16 Size=2048MB
Channel 1: LPDDR4, 50MHz
BW=32 Col=10 Bk=8 CS0 Row=16/15 CS=1 Die BW=16 Size=2048MB
256B stride
lpddr4_set_rate: change freq to 400000000 mhz 0, 1
lpddr4_set_rate: change freq to 800000000 mhz 1, 0
Trying to boot from BOOTROM
Returning to boot ROM...

U-Boot SPL 2021.10 (Oct 04 2021 - 15:09:26 +0000)
Trying to boot from MMC2
mmc_load_image_raw_sector: mmc block read error
Trying to boot from SPI
Trying to boot from MMC1
NOTICE:  BL31: v2.6(release):
NOTICE:  BL31: Built : 00:00:00, Jan  1 1980

Tow-Boot 2021.10 (Oct 04 2021 - 15:09:26 +0000)005 [variant: boot-installer]
[... 261 lines cut ...]
Error: Could not load font; falling back to built-in Roboto@22...       (/build/source/lvgui/hal.c #118)
libinput error: /nix/store/eeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee-libinput-aarch64-unknown-linux-gnu-1.16.4/share/libinput: failed to find data files
libinput error: Failed to load the device quirks from /nix/store/eeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee-libinput-aarch64-unknown-linux-gnu-1.16.4/share/libinput and /etc/libinput/local-overrides.quirks. This will negatively affect device behavior. [... cut ...]
libinput error: Failed to load the device quirks from /nix/store/eeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee-libinput-aarch64-unknown-linux-gnu-1.16.4/share/libinput and /etc/libinput/local-overrides.quirks. This will negatively affect device behavior. See https://wayland.freedesktop.org/libinput/doc/1.16.4/device-quirks.html for details.
xkbcommon: ERROR: failed to add default include path /nix/store/eeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee-xkeyboard-config-aarch64-unknown-linux-gnu-2.33/etc/X11/xkb
[ERROR] (xkbcommon_init): error: Could not create xkb_context.

Touch installer system
======================

Tow-Boot login:
```

</details>

paulbussler, 2025-01-25:
I think I am also experiencing this problem.  I flashed Tow-Boot 2023.07-007 to my SPI multiple times and always get the same thing, no display output while booting.