Tow-Boot/Tow-Boot #140: Possibly fix PinephonePro low batery startup problem

merlink01, 2022-03-28:
Is it possible to wait for a minimum battery level before continuing the boot process?
This will fix the boot loop problem the PPP has currently.

samueldr, 2022-03-28:
Yes, it is possible. Though it requires adding a driver for the PMIC to query the battery state.

I believe it was being worked on for upstreaming in U-Boot by other Pine64 community members.

suchipi, 2022-05-24:
Does anyone know of any resources on this issue (links where people might be working on it, etc)?

BillFleming, 2022-07-05:
This sounds like a basic feature that should be added to Tow-Boot.

samueldr, 2022-08-25:
#168 may help with part of the problem.

antoniprzybylik, 2022-09-22:
The "drained battery issue" is fixed in megi's u-boot. See this commit -- https://xff.cz/git/u-boot/commit/?h=ppp&id=7f8238fd608290152b143322178a5be21a447dc1

pinkuuu1, 2022-09-28:
> The "drained battery issue" is fixed in megi's u-boot. See this commit -- https://xff.cz/git/u-boot/commit/?h=ppp&id=7f8238fd608290152b143322178a5be21a447dc1

and that is exactly what is done in #168 pretty much. Just set the PMIC to take a higher current at start up, the other part of the fix is keeping the current limit set by the bootloader until the kernel knows more about the situation, so the kernel doesn't shut off the device from taking in any more charge and therefore creating this bootloop issue. Wthout the kernel patch, the PPP's PMIC Linux driver sets the input current to 0 at boot and keeps doing so until it has figured out more about the situation which is too late and the device has already shut off because the battery is dead :). Said fix was created by megi and is in most distros kernels now.

closed 2022-12-04 by merlink01

ArenM, 2023-08-13:
According to megi that fix isn't sufficient, unfortunately I don't have a PPP to test against, but I figured it was worth documenting.

From the #pinephone channel on irc.pine64.org
> \<megi\> it is not really fixed in Tow-Boot
> \<megi\> tow-boot uses this patch https://github.com/Tow-Boot/U-Boot/commit/465e5a1776d84f5e85836c61466b732c740ed8e5#diff-96512640d535c88213089e33fa98fc4a77b4f2bad8916e2cccc310cc24a43211 which raises current limit too high, raises it too late, and doesn't wait for battery charge to rise before booting
> \<megi\> here's a better approach https://megous.com/git/u-boot/commit/?h=ppp-2023.07&id=ced7bf1bc5f5539da35eeae0b7b70f0954871862