Tow-Boot/Tow-Boot #141: PineBook Pro, rescan usb causes reboot

ScarFez, 2022-03-28:
I noticed that when I do the Esc to bring up the bootable devices menu, and do the rescan usb; It reboots
Usually rescan usb is required for booting on any type of usb device. Although some times it is working without usb rescan. Seems to be intermittent... Magic ?

I do not have any logs and I do not have any way to connect something serial (uart) like to my pbp....
But I do have the latest version installed on SPI

rosbeef, 2022-09-18:
Related to #127

vincele, 2022-11-20:
Hello,
I cannot reproduce.
TB 2021.10-005 on sdcard.
emmc disconnected.
spi empty.

I tried plugging/unplugging an USB keyboard during the USB rescanning, or between ESC to get into the menu and the rescan. I got no reboot trying multiple times.

Do you also have this bug with the new version ?