Tow-Boot/Tow-Boot #142: An SHA/checksum would be nice to go along with the downloadable releases.

InfiniteOtter, 2022-04-04:
An SHA/checksum would be nice to go along with the downloadable releases. Since this is the first thing booted, it seems like making sure it is a secure and trustworthy version would be a good thing.

InfiniteOtter, 2022-04-11:
Just so you can check against my downloads SHA, I am putting this up. The naming convention for Tow boot makes one think the date the image was made is part of the naming convention, so that adds confusion. This was downloaded as the latest greatest on 4-11-2022
Name: Tow-Boot-release-2021.10-004.tar.gz
Size: 1876022 bytes (1832 KiB)
SHA256: 48eb837f7def4c61205c58ce7026e42648d58bd41c9b24bce9dd83eb81dd4515

PureTryOut, 2022-04-14:
> The naming convention for Tow boot makes one think the date the image was made is part of the naming convention

If you mean `2021.10-004`, then that largely comes from u-boot itself and is the version of u-boot itself. The `-004` indicates it's the 4th Tow-Boot release on that version of u-boot. I don't see how it's confusing to be honest, version numbers are always part of releases.

sifr01, 2024-07-23:
Please can this be actioned as it represents a long-standing security vulnerability. Dev signatures against build images would be preferable but checksums would be a gesture of goodwill.

For the meanwhile, downloaded today:
```
37882264 bytes
SHA512: f1183cd4f7c8036a5a9bca3f2f5fbc18c54f54fa627ca73ed9e80b3cf51d8e674eddc2bb1e188e65f70cc484e6a224251cbd4cf104bb667793474525e6a0f17d  pine64-pinephoneA64-2021.10-005.tar.xz
MD5: b166a84d85ce40bcd9597f731efedce6  pine64-pinephoneA64-2021.10-005.tar.xz
```