Tow-Boot/Tow-Boot #144: PineTab

alex-startengine, 2022-04-08:
Will you support the PineTab? It has the same SoC as PinePhone, the Allwinner A64. I would really appreciate it.

samueldr, 2022-04-08:
The lack of hardware to validate, and lack of community members with the hardware to validate, means I can't trivially do it. Though it should be relatively trivial to do the port, considering what is known about the PineTab.

So yes it should be possible, yes it would be desirable, but it requires someone that can validate, and ideally someone who knows how to get out of a *bad jam* in the unlikely event things go wrong.

YackyYam, 2022-04-10:
_**Raises hand.**_  I'll sacrifice mine to test.  You are talking about entering maskrom recovery and flashing from another computer when you talk about a "bad jam" or are you talking about soldering wires and using an external programmer like  raspberry pi? A external programmer like a MiniPro with an extension with a chip clamp would work also.

samueldr, 2022-04-10:
Software Jam, nothing hardware :)

YackyYam, 2022-04-10:
Welp I busted my digitizer last night while taking the back off or while it was off.  So call me a guinea pig!  Send me a DM or if you  want my discord that would work too.

PureTryOut, 2022-04-14:
@YackyYam Tow-Boot has a chat channel on Matrix which you can join, [#tow-boot:matrix.org](https://matrix.to/#/#tow-boot:matrix.org).

Danct12, 2022-04-26:
I can also help.. just need to get mine up and running again. I haven't turned it on for like a year now!

eighthourblink, 2022-04-27:
Willing to test as well. Just got my hands on a Pinetab and looking into all the options on what OS's work and whatnot

ivaylovassilev, 2022-09-24:
Hi, 
If you need I can also help.

Ivaylo

alxpettit, 2023-11-05:
I have several PineTab and would be happy to help test :)

zmy567, 2024-01-13:
Does this PineTab issue extend to the PineTab v2 or is this a separate issue to raise up on GitHub?

PR #341 (2025-01-17): Add PineTab support
Thank you @samueldr for your help with this!

Fixes #144
closed 2025-01-17 by samueldr