Tow-Boot/Tow-Boot #147: rockpi4c keyboard does not work in installer/early boot

elizagamedev, 2022-04-19:
I tried a number of different keyboards on all of the usb ports as well as trying through a USB hub. The installer was completely unresponsive.

I (maybe irresponsibly) flashed anyway by modifying the installer image to automatically flash at start. The issue persists when booting Linux (NixOS in my case), where the grub menu cannot be navigated since the keyboard does not work.

I don't have a serial cable, unfortunately, so I can't share the logs from my device.

Strit, 2022-04-20:
Same on rockpi4b for me. At least when booting the spi.installer.img flashed to SD card or eMMC.

CRTified, 2022-04-23:
This popped up a few times on the matrix channel, too, and IIRC some had luck with "other keyboards". 

I sadly don't know what the reason might be. While I've installed it using UART, I was able to navigate the menu with my keyboard when testing HDMI output.

CRTified, 2022-04-23:
One thing that I just remembered: Maybe try the other USB ports. There is at one port (top left, upper USB3.0) whose mode (Host vs. Device) is controlled by the small switch below the USB ports.

Strit, 2022-04-23:
Tried all USB ports, with the switch to right and left. Same outcome.

CRTified, 2022-04-23:
Strange, but thanks for reporting it.

elizagamedev, 2022-04-24:
Yeah, I tried all of the USB ports with pretty much every keyboard in my apartment. I really, really wanted to run a home server with rockpi4c + tow-boot + nixos but after a weekend of nothing but fiddling with it I couldn't get something that booted more than once.

NinjaTrappeur, 2022-05-17:
I have the same issue with my Rockpi4A. I also tried all the USB ports without success.

Tomaszal, 2022-07-14:
I am having the same issue with a few different keyboards. Right now my plan is to order keyboards from Amazon & return them if they don't work until I find one that works. Did anyone have any success getting a keyboard to work on Tow-Boot on RockPi4C at all? If so, which keyboard did you use?

jirutka, 2022-10-02:
I have the same problem on ROCK Pi 4 B+ (v1.73). :(

NinjaTrappeur, 2023-01-19:
I still have this issue with a ROCK Pi 4 A+ on the current development branch, ie. including #208. It seems like the `usb start` does not do the trick for that board.

NinjaTrappeur, 2023-01-19:
So, I went a bit deeper into this issue. I instrumented u-boot to print more logs about the keyboard initialization here: https://github.com/NinjaTrappeur/Tow-Boot/commit/b98e3935d3428d069b1d8aea59f3fdae0cf440f1

I got this at startup:

[image]

(sorry for the screen picture, I don't have the required hardware to connect to the uart console :()

It seems like we end up in this branch: https://github.com/u-boot/u-boot/blob/e092e3250270a1016c877da7bdd9384f14b1321e/common/usb_kbd.c#L550.

It seems like usb interruption submission `usb_int_msg(dev, data->intpipe, data->new, data->intpktsize, data->intinterval, false)` is failing.

I don't really know what to do with that information for now.

russnes, 2023-02-25:
Same issue here on a ROCK Pi 4 SE, two different keyboards all ports. It looks like @jirutka fixed it in a separate repo? How far away is it to merge the fix and have a new release?

NinjaTrappeur, 2023-02-26:
@russnes: the `usb start` preboot command has already been merged to the develop branch: https://github.com/Tow-Boot/Tow-Boot/blob/development/boards/radxa-RockPi4/default.nix#L21 (except it's the towboot infrastructure to perform the preboot command instead of a plain uboot patch).

As I mentioned above, it doesn't completely fix the issue for me.

samueldr, 2023-07-06:
Assuming this is fixed, if *this* issue as experience by individuals here is still present, please reopen/ask to reopen. Otherwise, if experiencing something vaguely similar, or really similar, don't hesitate to open a new issue with details!

closed 2023-07-06 by samueldr

caniko, 2025-03-25:
I just had the same issue happen to me on the rockpro64 @samueldr