Tow-Boot/Tow-Boot #151: [Request] ROCK64 Support

Absolucy, 2022-05-11:
Hi, I've been doing stuff with ARM64 EFI lately, and it'd be very useful if Tow-Boot supported the [ROCK64](https://www.pine64.org/devices/single-board-computers/rock64/) SBC, as it's what I currently have, and pretty much everything else is out-of-stock, so I have to rely on other people to test things for me, which can be kinda slow and inefficient at times.

The ROCK64 uses a RK3328 SoC (rather than the other PINE64 boards that use a RK3399 or Allwinner A64), and [seems to have support in mainline u-boot](https://github.com/u-boot/u-boot/blob/master/configs/rock64-rk3328_defconfig).

I'd try to add it myself, but I am struggling with the Nix-based build system and such, as it's very unfamiliar to me, and I can't get it to compile as the defconfig validation fails.

samueldr, 2022-05-11:
It should be possible, even close-to-trivial for a lot of it.

Though I don't have RK3328 hardware to validate against, so it makes things a lot more awkward when doing the port blind. In another place, I had discussions about the ROCK64, related to the SPI flash, and it looked like the platform firmware written to it is not assembled the same way as it is for RK3399.

So for the time being I can't commit to anything, but I'll try my hand at doing most of the leg work under the assumption that someone else (e.g. you) can validate it's working as expected.

Absolucy, 2022-05-12:
Yeah, I can test to make sure it works and such.

moosingin3space, 2022-05-23:
I'm having a look at this. I have a little bit of experience with u-boot and nix.