Tow-Boot/Tow-Boot #152: 2021.10-004 Pinebook Pro Boot Menu Always Shows

JRGonz, 2022-05-13:
Whenever a power cycle is performed the tow-boot boot menu shows without a timeout. Tow-Boot is on SPI. emmc has no OS and has 2 partitions (2GB swap, rest ext4 for storage only). There is nothing on it right now related to booting. SD is 512GB running Armbian. 

Ran into this issue a while back and it was not a legitimate issue until I upgraded the keyboard/trackpad fw and could not get past the menu requiring me to short pin6 to ground to bypass SPI and use U-Boot on SD.

JRGonz, 2022-05-13:
Just a note: Realized that if I wait a little while then choose reboot in the boot menu then the SD will boot. In my case I was stuck because I had no keyboard (tow-boot doesn't seem to support external keyboards for the boot menu?). Maybe this is just a SD initialize time issue?

vincele, 2022-11-20:
Hello, cannot reproduce this.
TB 2021.10-005 on sdcard.
emmc disconnected.
spi empty.

Did you also have this bug with the new version ?