Tow-Boot/Tow-Boot #153: 2021.10-004 Pinebook Pro No USB (External) Keyboard Support for Boot Menu

JRGonz, 2022-05-13:
When flashing my keyboard I was stuck at the Tow-Boot menu. Since I could not use the pbp keyboard at the time I had an external USB keyboard attached. The Tow-Boot boot menu did not seem to work with external keyboards. I tried three different keyboards (which worked in Armbian on the pbp).

Strit, 2022-07-23:
This seems to be the same issue as https://github.com/Tow-Boot/Tow-Boot/issues/147, since it's the same SoC and therefore mostly the same u-boot configuration.

vincele, 2022-11-20:
I can confirm, the problem is still there on 2021.10-005.

prokrypt, 2026-10-14:
I found the cause on a Pinebook Pro and have a fix (same applies to other boards with a built-in USB keyboard).

**Cause:** every USB keyboard registers as stdio device `usbkbd`. `usb_kbd_testc()` and `usb_kbd_getc()` don't use the device they're called with; they look it up again with `stdio_get_by_name(sdev->name)`, which always returns the *first* `usbkbd`. On the Pinebook Pro that's the internal keyboard, so an external keyboard is set up and registered (`coninfo` shows two `usbkbd`) but never read. Upstream U-Boot master still has this.

**Fix:** use `sdev->priv` directly in `testc`/`getc`, keep the poll timer per keyboard, name extra keyboards `usbkbd1`, `usbkbd2`, … and append them to `$stdin` (when it uses `usbkbd`), and remove each keyboard by its own name. Patch against `tb-2023.07-007`: https://github.com/prokrypt/pinebook-pro-towboot/blob/main/patches/0023-usb-kbd-multi-keyboard.patch

**Tested** on my Pinebook Pro: after "Rescan USB", the internal keyboard and a USB keyboard both work in the menu.

Two Pinebook Pro specifics: the right USB-A port (EHCI, behind an internal hub) works; the left USB-A port is the USB 3 (XHCI) controller, which has no interrupt-queue support in U-Boot, so a keyboard there is still rejected ("Failed to get keyboard state from device"). And with stock Tow-Boot the USB-A ports are scanned at power-on, so no rescan is needed there.

Prebuilt SPI image with this and the fix for #57 (touchpad freezing the keyboard): https://github.com/prokrypt/pinebook-pro-towboot/releases/tag/007-pbp2k