Tow-Boot/Tow-Boot #154: Add posibility to override environment settings in u-boot

AndreySV, 2022-05-13:
On PP it's not possible to boot kernel built with debug information using extlinux.conf (Mobian), because loaded images are too big.

```
mobian@mobian:~$ ls -lh /boot/ | grep test6
-rw-r--r-- 1 root root 198K Feb 17 18:10 config-5.15.13-suspend-fix-test6
drwxr-xr-x 3 root root 1.0K May 14 01:40 dtb-5.15.13-suspend-fix-test6
-rw-r--r-- 1 root root 100M May 14 01:42 initrd.img-5.15.13-suspend-fix-test6
-rw-r--r-- 1 root root 4.3M Feb 17 18:10 System.map-5.15.13-suspend-fix-test6
-rw-r--r-- 1 root root 7.3M Feb 17 18:10 vmlinuz-5.15.13-suspend-fix-test6
```
Boot log error:
```Enter choice: 1:	Mobian GNU/Linux 5.15.13-suspend-fix-test6
Retrieving file: /initrd.img-5.15.13-suspend-fix-test6
104826940 bytes read in 2267 ms (44.1 MiB/s)
Retrieving file: /vmlinuz-5.15.13-suspend-fix-test6
7605690 bytes read in 171 ms (42.4 MiB/s)
append: root=UUID=3d431641-b6f8-4dac-abb2-7dd84bf8d4ea console=ttyS0,115200 consoleblank=0 loglevel=7 rw splash plymouth.ignore-serial-consoles vt.global_cursor_default=0
Retrieving file: /dtb-5.15.13-suspend-fix-test6/allwinner/sun50i-a64-pinephone-1.2.dtb
37990 bytes read in 3 ms (12.1 MiB/s)
   Uncompressing Kernel Image
Moving Image from 0x40080000 to 0x40200000, end=4a550000
## Flattened Device Tree blob at 4fa00000
   Booting using the fdt blob at 0x4fa00000
ERROR: Failed to allocate 0x63f883c bytes below 0x4a000000.
ramdisk - allocation error
```
If 'bootm_size' variable is changed manually, then kernel successfully boots.
```
setenv bootm_size 0x1a000000
```
Unfortunately it's not possible to make this change persistent without recompiling tow-boot and the variable has to be change manually ever boot. It'd be nice, if it would be possible to load custom scripts before `scan_dev_for_extlinux` is executed.

samueldr, 2022-05-14:
If it's installed to SPI, `saveenv` or `env save` should work. Though note that it will also "break" in the future when the default environment is updated as it will prefer the saved environment to the built-in environment (#95).

Though updating these type of variables (offsets and sizes) is something that AFAIK is generally an accepted upstream change. A similar problem happened a few times in the past with the Raspberry Pi and NixOS.

And further, Tow-Boot should probably consider "owning" these variables and start harmonizing them between boards.

AndreySV, 2022-05-23:
I've tried boot.scr, but unfortunatelly it's tried after extlinux. So it will not help either, imho.

samueldr, 2022-05-23:
Noted, I hadn't taken the time to check that. And indeed *in the current situation* it won't help.

Main issue right now is "implementing" (mostly configuring) saved environment for mmcboot. The only "challenge" is that it shouldn't save anything to the main area of the mmc, so either it reserves the end of mmcboot0, or the end of mmcboot1 as the environment area. I guess reserving the end of mmcboot0 would be the least surprising choice, as it would act basically like SPI does.

AndreySV, 2022-05-25:
Thanks for the help. As a quick workaround, I've binary patched tow-boot image. I've changed command order to check for scripts first.
```
"run scan_dev_for_scripts; " 
"run scan_dev_for_extlinux; "
```
and created u-boot script in /boot/boot.scr with following content:
```
echo 'Run custom boot script'
setenv bootm_size 0x1a000000
run scan_dev_for_extlinux
```