Tow-Boot/Tow-Boot #164: Pinephone Pro UMS mode breaks and reboots

greendrake, 2022-07-21:
Booted into UMS mode, connected USB cable to a laptop and started cloning the eMMC filesystem. In the middle of the process the phone reboots itself. Any ideas why?

gjyoung1974, 2022-12-01:
same! it's very repeatable on my PPP. Happens every time I boot to UMS mode. after some period of time the device will just reboot

gjyoung1974, 2022-12-01:
seeing this in dmesg output

[ 5028.748376] usb 1-3.1.3.4.3: new full-speed USB device number 64 using xhci_hcd
[ 5030.037736] usb 1-3.1.3.4.3: unable to read config index 0 descriptor/start: -32
[ 5030.037744] usb 1-3.1.3.4.3: chopping to 0 config(s)
[ 5030.037749] usb 1-3.1.3.4.3: can't read configurations, error -32
[ 5030.220506] usb 1-3.1.3.4.3: new full-speed USB device number 65 using xhci_hcd
[ 5030.220651] usb 1-3.1.3.4.3: Device not responding to setup address.
[ 5030.428436] usb 1-3.1.3.4.3: Device not responding to setup address.
[ 5030.636389] usb 1-3.1.3.4.3: device not accepting address 65, error -71
[ 5030.636507] usb 1-3.1.3.4-port3: unable to enumerate USB device
[ 5035.716320] usb 1-3.1.3.4.3: new full-speed USB device number 66 using xhci_hcd
[ 5035.916353] usb 1-3.1.3.4.3: device descriptor read/64, error -32
[ 5036.228645] usb 1-3.1.3.4.3: device descriptor read/64, error -32
[ 5036.536341] usb 1-3.1.3.4.3: new full-speed USB device number 67 using xhci_hcd
[ 5036.736342] usb 1-3.1.3.4.3: device descriptor read/64, error -32
[ 5037.044381] usb 1-3.1.3.4.3: device descriptor read/64, error -32
[ 5037.156552] usb 1-3.1.3.4-port3: attempt power cycle
[ 5038.504462] usb 1-3.1.3.4.3: new full-speed USB device number 68 using xhci_hcd
[ 5038.534393] usb 1-3.1.3.4.3: unable to read config index 0 descriptor/start: -32
[ 5038.534398] usb 1-3.1.3.4.3: chopping to 0 config(s)
[ 5038.534401] usb 1-3.1.3.4.3: can't read configurations, error -32
[ 5038.732302] usb 1-3.1.3.4.3: new full-speed USB device number 69 using xhci_hcd
[ 5038.732402] usb 1-3.1.3.4.3: Device not responding to setup address.
[ 5038.944684] usb 1-3.1.3.4.3: Device not responding to setup address.
[ 5039.156347] usb 1-3.1.3.4.3: device not accepting address 69, error -71
[ 5039.156479] usb 1-3.1.3.4-port3: unable to enumerate USB device
[ 5066.668328] usb 1-3.1.3.4.3: new high-speed USB device number 70 using xhci_hcd
[ 5071.964485] usb 1-3.1.3.4.3: device descriptor read/64, error -110
[ 5078.456247] usb 1-3.1.3.4.3: new high-speed USB device number 71 using xhci_hcd
[ 5078.844291] usb 1-3.1.3.4-port3: attempt power cycle
[ 5080.004129] usb 1-3.1.3.4.3: new full-speed USB device number 72 using xhci_hcd
[ 5083.206146] usb 1-3.1.3.4.3: unable to read config index 0 descriptor/start: -32
[ 5083.206151] usb 1-3.1.3.4.3: chopping to 0 config(s)
[ 5083.206155] usb 1-3.1.3.4.3: can't read configurations, error -32
[ 5083.404080] usb 1-3.1.3.4.3: new full-speed USB device number 73 using xhci_hcd
[ 5083.404189] usb 1-3.1.3.4.3: Device not responding to setup address.
[ 5083.612227] usb 1-3.1.3.4.3: Device not responding to setup address.
[ 5083.820134] usb 1-3.1.3.4.3: device not accepting address 73, error -71
[ 5083.820343] usb 1-3.1.3.4-port3: unable to enumerate USB device
[ 5105.840010] usb 1-3.1.3.4.3: new high-speed USB device number 74 using xhci_hcd
[ 5106.652027] usb 1-3.1.3.4.3: device descriptor read/64, error -71
[ 5771.579691] perf: interrupt took too long (3168 > 3165), lowering kernel.perf_event_max_sample_rate to 63000
[ 6059.435507] usb 1-3.1.3.4.3: new high-speed USB device number 76 using xhci_hcd
[ 6059.640094] usb 1-3.1.3.4.3: New USB device found, idVendor=2207, idProduct=330c, bcdDevice= 2.23
[ 6059.640096] usb 1-3.1.3.4.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 6059.640098] usb 1-3.1.3.4.3: Product: Pinephone Pro
[ 6059.640099] usb 1-3.1.3.4.3: Manufacturer: Pine64
[ 6059.640099] usb 1-3.1.3.4.3: SerialNumber: 0ba81731f34e5977
[ 6059.641934] usb-storage 1-3.1.3.4.3:1.0: USB Mass Storage device detected
[ 6059.642042] scsi host1: usb-storage 1-3.1.3.4.3:1.0
[ 6060.133947] usb 1-3.1.3.4.3: USB disconnect, device number 76
[ 6086.326997] usb 1-3.1.3.4.3: new full-speed USB device number 78 using xhci_hcd
[ 6086.511096] usb 1-3.1.3.4.3: device descriptor read/64, error -32
[ 6086.807153] usb 1-3.1.3.4.3: Device not responding to setup address.
[ 6087.043285] usb 1-3.1.3.4.3: not running at top speed; connect to a high speed hub
[ 6087.044455] usb 1-3.1.3.4.3: New USB device found, idVendor=2207, idProduct=330c, bcdDevice= 2.23
[ 6087.044458] usb 1-3.1.3.4.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 6087.044461] usb 1-3.1.3.4.3: Product: Pinephone Pro
[ 6087.044462] usb 1-3.1.3.4.3: Manufacturer: Pine64
[ 6087.044464] usb 1-3.1.3.4.3: SerialNumber: 0ba81731f34e5977
[ 6087.051181] usb-storage 1-3.1.3.4.3:1.0: USB Mass Storage device detected
[ 6087.051622] scsi host1: usb-storage 1-3.1.3.4.3:1.0
[ 6088.064256] scsi 1:0:0:0: Direct-Access     Linux    UMS disk 0       ffff PQ: 0 ANSI: 2
[ 6088.064853] sd 1:0:0:0: Attached scsi generic sg1 type 0
[ 6088.065622] sd 1:0:0:0: [sdb] 241664000 512-byte logical blocks: (124 GB/115 GiB)
[ 6088.066582] sd 1:0:0:0: [sdb] Write Protect is off
[ 6088.066587] sd 1:0:0:0: [sdb] Mode Sense: 0f 00 00 00
[ 6088.067361] sd 1:0:0:0: [sdb] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[ 6088.160030] Alternate GPT is invalid, using primary GPT.
[ 6088.160039]  sdb: sdb1 sdb2
[ 6088.201542] sd 1:0:0:0: [sdb] Attached SCSI removable disk
[ 6112.101912] usb 1-3.1.3.4.3: USB disconnect, device number 78
[ 6112.103318] sd 1:0:0:0: [sdb] Synchronizing SCSI cache
[ 6112.103395] sd 1:0:0:0: [sdb] Synchronize Cache(10) failed: Result: hostbyte=DID_NO_CONNECT driverbyte=DRIVER_OK

gjyoung1974, 2022-12-07:
Disregard. My issue was due to a bad USB-C cable

greendrake, 2022-12-08:
> Disregard. My issue was due to a bad USB-C cable

So, you consider it not a bug that a bad cable was causing PPP to reboot?