Tow-Boot/Tow-Boot #166: Request for a possible/feasible graphical UI.(like Boot menu)

Alexyo21, 2022-07-29:
Good morning, sorry for bothering you, I know that what's written in the title can be a little hard to make because of the uboot that doesn't initialise the display, but I was wondering that if it maybe possible to make, it would be a great feature. Thanks a lot.
Have a good day.

samueldr, 2022-07-29:
> because of the uboot that doesn't initialise the display

It's the only reason it's not already done :). It is desirable and wanted already.

Thanks for the kind words!

Alexyo21, 2022-07-30:
Well, I answered to myself in the end 😂😅
I was thinking, I noticed that on pinebook pro(the one with the same socket as pinephone pro) has a display gui menu [image]
As I read here I dunno for sure.
Is it right?

samueldr, 2022-07-30:
Yes. Any targets with display out support (this should probably be documented in the device pages) can show that boot selection menu.

Devices without display out *technically* can use that menu, but requires serial use for it to be useful.

* * *

>  I noticed that on pinebook pro(the one with the same socket (about) as pinephone pro) has a display gui menu

While the Pinephone Pro and the Pinebook Pro share basically the same SoC, they don't share the same hardware details. At a high level, the Pinebook Pro uses an eDP display, while the Pinephone Pro uses a MIPI display. The current driver in U-Boot for RK3399 (IIRC) only supports eDP, DP and HDMI.

Alexyo21, 2022-07-31:
> Yes. Any targets with display out support (this should probably be documented in the device pages) can show that boot selection menu.
>
>
>
> Devices without display out *technically* can use that menu, [...]

Ok now I understood, so in the end the if we use a serial console we can interact with it.
But the great question is: 
Will the future driver update support MIPI display ?
So we won't need serial to use it.
Tell me if I said something wrong.
Thanks you again.

samueldr, 2022-07-31:
> Will the future driver update support MIPI display ?

AFAIK no one is currently working on a MIPI driver for RK3399 for U-Boot.

Alexyo21, 2022-07-31:
> > Will the future driver update support MIPI display ?
> 
> 
> 
> AFAIK no one is currently working on a MIPI driver for RK3399 for U-Boot.

Good to know, can you tell me more for example, in case I want to begin the development of this driver, where do I need to go to get the source code about the RK3399 driver and information about MIPI, just to set a point to begin with? Thank you very much.