Tow-Boot/Tow-Boot #167: pine64-pinephoneA64: battery is not charging

mkbel, 2022-07-31:
Affected version [release 2021.10-004](https://github.com/Tow-Boot/Tow-Boot/commit/d7f7cf1ac3432ca6dc0287dc0893b5c6727bd6d6)

It is not possible to boot into mobian-installer-pinephone-phosh-20220717.img no matter if it is on SD card or internal storage if the battery is discharged.

Steps to reproduce:
1. Phone battery is completely discharged. (Voltage measured about 2.5V.) There is mobian-installer-pinephone-phosh-20220717.img burned to e-mmc or SD card.
2. Connect USB-C 3A 5V charger to phone.
3. Phone is turned on by the charger.
Expected:
Tow-boot starts charging the battery during boot process.
Actual:
There is endless boot loop switching between tow-boot and mobian splash screen. There is no way to charge phone battery from tow-boot and mobian installer resets the device immediately after displaying splash screen. Phone battery is not charging during this process, so this boot-loop is endless.

Pressing volume-up to start jump-drive also doesn't charge the battery. Phone is not rebooting, its storage can be accessed from computer but the battery is also not charging.

Possible workaround:
Removing battery from the phone, charging it little bit with laboratory power supply and returning it back to the phone solved the problem. If the battery is at least partially charged Mobian installer can start successfully.

Strit, 2022-08-01:
The Pinephone not being able to boot when the battery is completely depleted is not a u-boot issue as far as I know.

Charging in this case can take hours before it gets to a point where it can actually boot fully.

mkbel, 2022-08-02:
> The Pinephone not being able to boot when the battery is completely depleted is not a u-boot issue as far as I know.

I can agree with that.

> Charging in this case can take hours before it gets to a point where it can actually boot fully.

I don't agree with that.
1. If there is no firmware interaction with charging circuitry there is no charging at all. (I am basing this claim on the fact, that PINE64 is distributing firmware image [pine64-pinephone-charging.img.xz](https://github.com/Tow-Boot/Tow-Boot/files/9244746/pine64-pinephone-charging.img.xz.zip) for the sole purpose of charging the battery. If there would be some charging without any firmware interaction mere existence of that firmware would make no sense.)
2. I suspect that there is no firmware interaction with charging circuitry in Tow-Boot. (I am basing this claim on the fact, that after about 30 minutes in jump-drive mode, battery was still at its cut-off voltage of 2.5V.)
3. So in case 2. is true "charging" would take infinite time to get to a point where it can actually boot fully.
4. On the other hand charging battery at 1000mA for 5 minutes was sufficient to get to a point where it can actually boot fully.
5. In hypothetical case Tow-Boot would detect 3A power supply and charge phone battery at its maximum of 1500mA 3.5 minutes would be sufficient to get to a point where it can actually boot fully.

**How Tow-Boot can be improved:**

1. Start charging the battery as part of the boot process.
This would be ideal but I don't expect this to be implemented. I expect that there is no USB handshake in the basic part of the boot process. So power supply limit is unknown. So adding this would be too much effort for benefiting only a couple of users.

2. Charge the battery in jump-drive mode.
In this mode some USB handshake is for sure done. So power delivery capability of USB port is already known or adding it would be small effort. Even if not all cases would be handled (e.g. not detecting dedicated charging port) and charging the battery with some small conservative current in order of hundreds of mA would be very beneficial because of long time spent in this mode. (Flashing OS image takes tens of minutes. User can leave the device in jump-drive mode intentionally much longer.)

3. Do not let the battery fully discharge if there is no OS to start. Power off the device in case of any boot process error.
It is my mere speculation that my battery discharged because I installed Tow-Boot and then flashed OS image one day later. PinePhone is powered on by connecting power supply. So I speculate, that at some point I connected the charger to fully charge the phone and disconnected it later. If Tow-Boot doesn't power off on error battery was discharged while Tow-Boot was trying to boot non-existent OS. This feature is implemented in [p-boot](https://xnux.eu/p-boot/)

megous, 2022-08-20:
Yeah, bootloader must not let the system boot or do any power hungry tasks until there's some small charge in the battery. This is normal feature on iOS and I guess on android phones, too. If you have battery completely diuscharged on iOS, iOS will not allow you to boot to actual OS, it will just show the low battery screen, and wait for several minutes of charging and then continue with the boot.

This should be done by Tow-Boot, too.

khimaros, 2022-11-02:
i believe this may now be working as described.

i started with a completely unbootable phone due to dead battery

i powered on while holding volume up, left for a few minutes, then powered off

when i powered back on, it booted successfully.

maybe this issue can be closed and the behavior documented?

AndreySV, 2023-07-29:
Android phones don't boot strait into OS as well, when the battery is discharged. Usually there are couple precharge loops:
1. with minimal enabled consumers (no display, no backlight, no user feedback)
2. with display support to get user feedback, that device is alive, but completely discharged.
After device is charged enough, full OS is loaded in charging mode usually (MTK, QCOM).

It would be nice to have charger enabled in bootloader, like it was done for Librem 5:
https://github.com/u-boot/u-boot/blob/master/board/purism/librem5/librem5.c#L354

Currently AXP803 isn't supported by u-boot. I've seen several attempts to get it merged, but nothing was merged.

1. https://patchwork.ozlabs.org/project/uboot/list/?series=&submitter=&state=*&q=AXP803&archive=both&delegate=
2. https://build.opensuse.org/package/view_file/home:duwe:Teres-I/u-boot/0103-Add-AXP803-PMIC-support.patch?expand=1

There is some basic support for AXP803 in TF-A and in crust. I'm not sure, whether crust and TF-A will make adding charger support in AXP803 in u-boot more complicated? @samueldr should all accesses to AXP803 go through crust?

AndreySV, 2023-07-29:
As a workaround for that I'm using Tow-Boot's feature to boot from SD-card. My main system is installed to eMMC and SD-card contains content of pine64-pinephone-charging.img.

When PinePhone is discharged
1. Plug PP into charger
2. Phone turns on
3. Press Volume Down to boot from SD card
4. Wait until Phone is charged a bit and reboot into main system

```
wget https://github.com/dreemurrs-embedded/Jumpdrive/releases/download/0.8/pine64-pinephone.img.xz
aunpack pine64-pinephone.img.xz
truncate -s 60M pine64-pinephone-charging.img
sudo kpartx -v -a pine64-pinephone-charging.img
mkdir charging
sudo mount /dev/mapper/loop0p1 charging
SRC=$PWD/charging

mkdir /tmp/sd_charging
sudo mount /dev/mmcblk0p1 /tmp/sd_charging
cd /tmp/sd_charging
mkdir -p dtb-jumpdrive-charging/allwinner
cp -v $SRC/sun50i-a64-pinephone-1.* dtb-jumpdrive-charging/allwinner/
cp $SRC/initramfs.gz initrd.img-jumpdrive-charging
cp $SRC/Image.gz vmlinuz-jumpdrive-charging
mkdir extlinux
vi extlinux/extlinux.conf

cd $SRC/..
sudo umount $SRC
sudo umount /tmp/sd_charging
sudo kpartx -d pine64-pinephone-charging.img
```

extlinux.conf content
```
default l0
menu title U-Boot menu
prompt 0
timeout 10

label l0
	menu label Jumpdrive PinePhone charging 
	linux /vmlinuz-jumpdrive-charging
	initrd /initrd.img-jumpdrive-charging
	fdtdir /dtb-jumpdrive-charging/
	append console=ttyS0,115200 loglevel=8 
```